espressif/esp-idf #3192: mbedtls_ssl_handshake errors specifying failed to open new connection

AadiMehta, 2019-03-17:
Description:

I am trying to send log file to the local server using esp_http_client. I am using mdns service to identify local server in the network(This part seems to be working fine). We are also using mqtt for other functionality. 

9 out of 10 times the file sending succeeds but when failed, even after trying multiple times it doesn't succeeds. Logs for the ssl_handshake errors are following:

```
I (212340) ESP32Project.utils.networking: DATA={"type": "REQUEST", "data": "", "msg_id": "0x240ac40f4eec_29500403", "command": "health"}
I (212355) ESP32Project.utils.networking: File size to send: 2655
I (219817) wifi: bcn_timout,ap_probe_send_start
E (220818) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (220819) esp-tls: Certificate verified.
E (220822) esp-tls: Failed to open new connection
E (220824) TRANS_SSL: Failed to open a new connection
E (220830) HTTP_CLIENT: Connection failed, sock < 0
E (220835) ESP32Project.utils.networking: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
I (220850) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=37089
I (220855) ESP32Project.utils.networking: DATA={"type": "REQUEST", "data": "", "msg_id": "0x240ac40f4eec_29500403", "command": "health"}
I (220870) ESP32Project.utils.networking: File size to send: 2881
E (228061) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (228062) esp-tls: Certificate verified.
E (228064) esp-tls: Failed to open new connection
E (228067) TRANS_SSL: Failed to open a new connection
E (228073) HTTP_CLIENT: Connection failed, sock < 0
E (228078) ESP32Project.utils.networking: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
I (228093) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=13340
I (228096) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=12092
I (230500) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=30349
I (464530) ESP32Project.utils.networking: DATA={"data": "", "msg_id": "0x240ac40f4eec_83120411", "type": "REQUEST", "command": "chipInfo"}
I (464552) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=53744
I (758081) ESP32Project.utils.networking: DATA={"type": "REQUEST", "data": "", "msg_id": "0x240ac40f4eec_29500403", "command": "health"}
I (758096) ESP32Project.utils.networking: File size to send: 3501
E (758704) esp-tls: mbedtls_ssl_handshake returned -0x10
I (758704) esp-tls: Certificate verified.
E (758707) esp-tls: Failed to open new connection
E (758709) TRANS_SSL: Failed to open a new connection
E (758715) HTTP_CLIENT: Connection failed, sock < 0
E (758720) ESP32Project.utils.networking: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
I (758735) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=31980
I (758741) ESP32Project.utils.networking: MQTT_EVENT_PUBLISHED, msg_id=26221
I (770720) ESP32Project.utils.networking: DATA={"type": "REQUEST", "data": "", "msg_id": "0x240ac40f4eec_29500403", "command": "health"}
I (770737) ESP32Project.utils.networking: File size to send: 3810
E (771331) esp-tls: mbedtls_ssl_handshake returned -0x10
```

Note: It does say that the certificate has been verified, so that means mdns is working as expected and esp was able to contact the local server and handshake was initiated, but then it fails saying fail to open a new connection.

AadiMehta, 2019-04-07:
Any update on this?

AadiMehta, 2019-04-08:
After turning on mbedtls debug I found out the issue to be at:
mbedtls_ecdh_make_public() function
```
I (194091) mbedtls: ssl_tls.c:2763 <= flush output

I (194097) mbedtls: ssl_cli.c:2851 => write client key exchange

W (194549) mbedtls: ssl_cli.c:2912 mbedtls_ecdh_make_public() returned -16 (-0x0010)

I (194549) mbedtls: ssl_tls.c:8031 <= handshake

E (194552) esp-tls: mbedtls_ssl_handshake returned -0x10
I (194558) esp-tls: Certificate verified.
E (194565) esp-tls: Failed to open new connection
E (194568) TRANS_SSL: Failed to open a new connection
E (194573) HTTP_CLIENT: Connection failed, sock < 0
E (194579) ESP32Project.utils.networking: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
```
NOTE: HTTPS requests fail only when I am trying to send some attachment to the server. Otherwise, all other HTTPS API requests are passing to the same server.

Also when I deselect ecdh ciphers from the mbedtls config, issue is no more there. I have tried enabling ecdh ciphersuite explicitly on the apache server but to no success. Please let me know if the findings are correct.

jitin17, 2019-04-09:
@AadiMehta Can you give us some information about stack and heap consumption when you are sending some attachments with ecdh ciphers enabled?

jitin17, 2019-04-09:
@AadiMehta This looks like a memory issue. The fact that it works on disabling ECDH reinforces the claim. Here's a similar issue.
https://tls.mbed.org/discussions/bug-report-issues/mbedtls_ecdh_make_public-returned-16
Could you call esp_get_minimum_free_heap_size() in your code and respond with the heap size during the SSL handshake?

lexus2k, 2019-05-14:
Hi All,

Definitely, I see the same issue: tls returns "0x4c" error.  (I use official ESP-IDF v3.2)

Sometimes tls connection opens, but mostly it fails. From my observations, the issue happens at my home network (Asus RT-66), but I see it very rarely at the office (I believe we have some MIkrotiks). So, maybe, it relates to some network conditions. And the root cause is NOT memory issue - I logged min free heap size:
```
I (13528) WEB: Validating version per https://github.com/lexus2k/nixie_clock/raw/master/binaries/nixie_clock.txt
I (13548) EVENT: MIN STACK: 169412
I (16128) EVENT: MIN STACK: 122340
I (18708) EVENT: MIN STACK: 119596
I (21288) EVENT: MIN STACK: 119596
E (21418) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (21418) esp-tls: Failed to open new connection
E (21418) TRANS_SSL: Failed to open a new connection
E (21418) HTTP_CLIENT: Connection failed, sock < 0
E (21428) WEB: Failed to perform HTTP GET request from https://github.com/lexus2k/nixie_clock/raw/master/binaries/nixie_clock.txt
I (23868) EVENT: MIN STACK: 119596
```

jitin17, 2019-05-30:
@lexus2k Your error condition is `-0x4c` which is `MBEDTLS_ERR_NET_RECV_FAILED`, and this is different from the error code `-0x10` seen in the above-mentioned problem. `MBEDTLS_ERR_NET_RECV_FAILED` mostly implies some network issues. It would be helpful if you can enable `mbedtls` logging and do a pcap capture.

lexus2k, 2019-05-30:
@jitin17 here are the logs, you requested:

```
I (18013) WEB: Validating version per https://github.com/lexus2k/nixie_clock/raw/master/binaries/nixie_clock.txt
I (18873) mbedtls: ssl_tls.c:8021 => handshake

I (18873) mbedtls: ssl_cli.c:3405 client state: 0

I (18873) mbedtls: ssl_tls.c:2751 => flush output

I (18883) mbedtls: ssl_tls.c:2763 <= flush output

I (18883) mbedtls: ssl_cli.c:3405 client state: 1

I (18893) mbedtls: ssl_tls.c:2751 => flush output

I (18893) mbedtls: ssl_tls.c:2763 <= flush output

I (18903) mbedtls: ssl_cli.c:774 => write client hello

I (18913) mbedtls: ssl_tls.c:3180 => write handshake message

I (18913) mbedtls: ssl_tls.c:3337 => write record

I (18923) mbedtls: ssl_tls.c:2751 => flush output

I (18923) mbedtls: ssl_tls.c:2770 message length: 239, out_left: 239

I (18933) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 239 (-0xffffff11)

I (18943) mbedtls: ssl_tls.c:2803 <= flush output

I (18943) mbedtls: ssl_tls.c:3470 <= write record

I (18953) mbedtls: ssl_tls.c:3314 <= write handshake message

I (18953) mbedtls: ssl_cli.c:1106 <= write client hello

I (18963) mbedtls: ssl_cli.c:3405 client state: 2

I (18973) mbedtls: ssl_tls.c:2751 => flush output

I (18973) mbedtls: ssl_tls.c:2763 <= flush output

I (18983) mbedtls: ssl_cli.c:1499 => parse server hello

I (18983) mbedtls: ssl_tls.c:4305 => read record

I (18993) mbedtls: ssl_tls.c:2532 => fetch input

I (18993) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (19433) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (19433) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (19433) mbedtls: ssl_tls.c:2738 <= fetch input

I (19443) mbedtls: ssl_tls.c:2532 => fetch input

I (19443) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 102

I (19453) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 102

I (19453) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 97 (-0xffffff9f)

I (19463) mbedtls: ssl_tls.c:2738 <= fetch input

I (19473) mbedtls: ssl_tls.c:4379 <= read record

[... 512 lines cut ...]
I (31033) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 3187

I (31043) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 3187

I (31043) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 1353 (-0xfffffab7)

I (31413) EVENT: MIN HEAP: 119424, CURRENT HEAP: 133232
I (36053) mbedtls: ssl_tls.c:2717 in_left: 1358, nb_want: 3187

I (36053) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned -76 (-0x004c)

W (36053) mbedtls: ssl_tls.c:5019 mbedtls_ssl_fetch_input() returned -76 (-0x004c)

W (36063) mbedtls: ssl_tls.c:4338 ssl_get_next_record() returned -76 (-0x004c)

W (36073) mbedtls: ssl_tls.c:5480 mbedtls_ssl_read_record() returned -76 (-0x004c)

I (36073) mbedtls: ssl_tls.c:8031 <= handshake

E (36083) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (36093) esp-tls: Failed to open new connection
E (36093) TRANS_SSL: Failed to open a new connection
E (36103) HTTP_CLIENT: Connection failed, sock < 0
E (36103) WEB: Failed to perform HTTP GET request from https://github.com/lexus2k/nixie_clock/raw/master/binaries/nixie_clock.txt
```

jitin17, 2019-05-31:
@lexus2k I think somehow there seems to be a problem in reading the certificate received from `https://raw.githubusercontent.com/lexus2k/nixie_clock/master/binaries/nixie_clock.txt`

This is the part of the log where the problem appears.
```
I (30983) mbedtls: ssl_tls.c:5440 => parse certificate

I (30993) mbedtls: ssl_tls.c:4305 => read record

I (30993) mbedtls: ssl_tls.c:2532 => fetch input

I (31003) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (31003) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (31013) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (31023) mbedtls: ssl_tls.c:2738 <= fetch input

I (31023) mbedtls: ssl_tls.c:2532 => fetch input

I (31033) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 3187

I (31043) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 3187

I (31043) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 1353 (-0xfffffab7)

I (31413) EVENT: MIN HEAP: 119424, CURRENT HEAP: 133232

W (36053) mbedtls: ssl_tls.c:5019 mbedtls_ssl_fetch_input() returned -76 (-0x004c)

W (36063) mbedtls: ssl_tls.c:4338 ssl_get_next_record() returned -76 (-0x004c)

W (36073) mbedtls: ssl_tls.c:5480 mbedtls_ssl_read_record() returned -76 (-0x004c)
```
Somehow the entire 3187 bytes of certificate is not being read and `-0x004c` implies system error.
```
I (36053) mbedtls: ssl_tls.c:2717 in_left: 1358, nb_want: 3187

I (36053) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned -76 (-0x004c)
```
Since the problem is random, I think pcap capture would be of great help here.

On a side note, I tried to connect to the same URL with `esp_http_client` (ESP-IDF v3.2) and faced no issue. I see the entire 3187 bytes of the certificate being read on my device.

```
I (10883) mbedtls: ssl_tls.c:5440 => parse certificate

I (10893) mbedtls: ssl_tls.c:4305 => read record

I (10893) mbedtls: ssl_tls.c:2532 => fetch input

I (10903) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (10913) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (10913) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (10923) mbedtls: ssl_tls.c:2738 <= fetch input

D (10933) mbedtls: ssl_tls.c:4050 input record: msgtype = 22, version = [3:3], msglen = 3182

I (10933) mbedtls: ssl_tls.c:2532 => fetch input

I (10943) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 3187

I (10953) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 3187

I (10953) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 3182 (-0xfffff392)

I (10963) mbedtls: ssl_tls.c:2738 <= fetch input

D (10993) mbedtls: ssl_tls.c:3620 handshake message: msglen = 3182, type = 11, hslen = 3182

I (10993) mbedtls: ssl_tls.c:4379 <= read record
```

lexus2k, 2019-05-31:
@jitin17  Thank you for clarifications. I understand that the problem relates to network issue. And what I have for now:
1. ESP32 experiences TLS issues at my home network
2. ESP32 has no issues with TLS at the office network
3. Other home devices have no issues with SSL connection to github site.

Thus, I came to conclusion, that ESP32 has some specific code in TLS implementation, which is sensitive to network environment. 
How can I get PCAP logs? ASUS RT-66 doesn't allow to get such logs, and the router is the only device at my place between ESP32 and github server. Of course, I can try to direct TCP traffic from ESP32 via hotspot on the laptop, but that will be different case.

jitin17, 2019-06-03:
@lexus2k You can get PCAP logs with the help of Wireshark, TCPDump etc.

lexus2k, 2019-06-04:
@jitin17 Hi, I know about Wireshark and TCPDump, but as I said above, to capture TCP traffic the only place to run these tools is WiFi router, and my router doesn't allow to run these tools.
Anyway, I tried to reproduce issue again several times, and everything works OK for now. I did no changes to esp32 firmware, maybe the issue was caused by network provider. 
I will keep an eye on TLS behavior, and will report as soon as I meet the problem again.

jitin17, 2019-06-06:
@lexus2k Glad to know it got fixed. 

@AadiMehta I am closing this issue, please feel free to reopen it if you still face a similar problem.

closed 2019-06-06 by jitin17

ankitdaf, 2019-07-02:
We are facing the same problem. 

A small snippet of the log is here : 

E (4396500) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (4396500) esp-tls: Failed to open new connection
E (4396500) TRANS_SSL: Failed to open a new connection
E (4396500) HTTP_CLIENT: Connection failed, sock < 0
E (4396510) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT

cheezum, 2019-08-12:
Same problem here (-0x4c) after updating. 
I'm running on 4.0-dev-1443 with CMake as compiler.
I tried to enable the debugging, but that gives a lot of errors on the OTA partition and partitions not being bootable.

mahavirj, 2019-08-12:
@ankitdaf @cheezum We have fix for this regression, it will reflect on github repository soon. Until then please try attached fix, [mbedtls_fix_errno_handling.tar.gz](https://github.com/espressif/esp-idf/files/3492911/mbedtls_fix_errno_handling.tar.gz)

cheezum, 2019-08-13:
Thanks. This patch works for me !!

Bertoleti, 2019-11-27:
I have the same issue.
I'm using IDF 4.13 on Eclipse. How can I install this patch with corrections?

E (12819080) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4c
E (12819080) esp-tls: Failed to open new connection
E (12819090) TRANS_SSL: Failed to open a new connection
I (12819090) wifi station: retry to connect to the AP
E (12819090) HTTP_CLIENT: Connection failed, sock < 0
I (12819100) wifi station: connect to the AP fail
I (12819100) http events: HTTP_EVENT_DISCONNECTED
I (12821680) tcpip_adapter: sta ip: 10.211.6.189, mask: 255.255.248.0, gw: 10.211.0.1
I (12821680) wifi station: got ip:10.211.6.189

MohdAzka, 2020-12-23:
Hi 
I was working on esp-idf  example code 'simple_ota_example' and i was getting the same error like that "Failed to open new connection" 

E (3893) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7200[0m
I (3893) esp-tls-mbedtls: Certificate verified.[0m
E (3893) esp-tls: Failed to open new connection[0m
E (3903) TRANS_SSL: Failed to open a new connection[0m
E (3913) HTTP_CLIENT: Connection failed, sock < 0[0m
E (3913) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT[0m
E (3923) esp_https_ota: Failed to establish HTTP connection[0m
E (3923) simple_ota_example: Firmware upgrade failed[0m

until i have change my url from "https://192.168.1.141/..." to "http://192.168.1.141/..."

now everything is working fine

lexus2k, 2020-12-24:
Hi @MohdAzka 

Everything is find for me now, I don't observe the issue, since I switched to newer IDF releases. Both v4.0.1 and v3.3 are good for me.