espressif/esp-idf #3194: Multiple unused warnings in BT stack

no1seman, 2019-03-18:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version : [v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version : v4.0-dev-71-g5beb2802e
- Build System:         [Make & CMake]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB | external 5V]

## Problem Description
Multiple unused variables warnings in BT stack:
.......
[560/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/btc/core/btc_dm.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/btc/core/btc_dm.c: In function 'btc_dm_ble_auth_cmpl_evt':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/btc/core/btc_dm.c:251:17: warning: variable 'status' set but not used [-Wunused-but-set-variable]
     bt_status_t status = BT_STATUS_FAIL;
                 ^
[561/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/btc/core/btc_ble_storage.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/btc/core/btc_ble_storage.c: In function '_btc_read_le_key':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/btc/core/btc_ble_storage.c:753:18: warning: unused variable 'bd_str' [-Wunused-variable]
             char bd_str[20] = {0};
                  ^
[580/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/device/interop.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/device/interop.c: In function 'interop_match':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/device/interop.c:50:18: warning: unused variable 'bdstr' [-Wunused-variable]
             char bdstr[20] = {0};
                  ^
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/device/interop.c: At top level:
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/device/interop.c:30:20: warning: 'interop_feature_string' defined but not used [-Wunused-function]
 static const char *interop_feature_string(const interop_feature_t feature)
                    ^
[608/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/hci/hci_hal_h4.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c: In function 'hci_hal_h4_hdl_rx_packet':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:280:17: warning: variable 'len' set but not used [-Wunused-but-set-variable]
         uint8_t len = 0;
                 ^
[663/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/btm/btm_ble_gap.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c: In function 'BTM_VendorHciEchoCmdCallback':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c:763:21: warning: variable 'echo' set but not used [-Wunused-but-set-variable]
     uint8_t status, echo;
                     ^
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c:763:13: warning: variable 'status' set but not used [-Wunused-but-set-variable]
     uint8_t status, echo;
             ^
[667/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/btm/btm_pm.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_pm.c:953:20: warning: 'mode_to_string' defined but not used [-Wunused-function]
 static const char *mode_to_string(tBTM_PM_MODE mode)
                    ^
[670/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/gatt/gatt_attr.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_attr.c: In function 'gatt_profile_db_init':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_attr.c:339:23: warning: variable 'status' set but not used [-Wunused-but-set-variable]
     tGATT_STATUS      status;
                       ^
[671/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/gatt/gatt_api.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c: In function 'GATT_SetIdleTimeout':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c:1158:21: warning: variable 'status' set but not used [-Wunused-but-set-variable]
     BOOLEAN         status = FALSE;
                     ^
[674/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/gap/gap_ble.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gap/gap_ble.c: In function 'gap_attr_db_init':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gap/gap_ble.c:377:21: warning: variable 'status' set but not used [-Wunused-but-set-variable]
     tGATT_STATUS    status;
                     ^
[681/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/btm/btm_sec.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_sec.c: In function 'btm_sec_bond_by_transport':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/btm/btm_sec.c:952:23: warning: variable 'p_features' set but not used [-Wunused-but-set-variable]
     UINT8            *p_features;
                       ^
[693/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/gatt/gatt_utils.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c: In function 'gatt_end_operation':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c:2143:25: warning: variable 'operation' set but not used [-Wunused-but-set-variable]
     UINT8               operation;
                         ^
[695/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/l2cap/l2c_main.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c: In function 'l2c_rcv_acl_data':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:126:37: warning: variable 'psm' set but not used [-Wunused-but-set-variable]
     UINT16      l2cap_len, rcv_cid, psm;
                                     ^
[715/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/smp/smp_api.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_api.c: In function 'SMP_CreateLocalSecureConnectionsOobData':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_api.c:589:14: warning: variable 'bd_addr' set but not used [-Wunused-but-set-variable]
     UINT8   *bd_addr;
              ^
[723/1069] Building C object esp-idf/bt/CMakeFiles/idf_component_bt.dir/bluedroid/stack/smp/smp_keys.c.obj
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_keys.c: In function 'smp_calculate_local_commitment':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_keys.c:1156:81: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
                                     for Just Works/Numeric Comparison models\n");
                                                                                 ^
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_keys.c: In function 'smp_calculate_peer_commitment':
/home/user/esp/esp-mdf/esp-idf/components/bt/bluedroid/stack/smp/smp_keys.c:1201:61: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
                 for Just Works/Numeric Comparison models\n");
                                                             ^

### Expected Behavior
No any warnings
### Actual Behavior
Multiple warnings, see above log.
### Steps to repropduce
)

Weijian-Espressif, 2019-03-18:
@no1seman , You disable `Disable BT debug logs (minimize bin size) ` in menuconfig, am i right?
I will fix the issue as soon as possible.

no1seman, 2019-03-18:
@Weijian-Espressif,  yes, you are rigth, BT debug logs disabled in sdkconfig.
thanks in advance

Weijian-Espressif, 2019-03-18:
@no1seman , I have created a merge request to fix this bug, it can be merged into IDF this week.

Alvin1Zhang, 2019-03-21:
Will keep this issue open until the fix is merged, thanks.

closed 2019-03-21 by Alvin1Zhang