espressif/esp-idf #3195: [WIN10/CMake] Issues compiling on Windows 10

souryavarenya, 2019-03-18:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0-dev-71-g5beb2802e
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a5 - 5.2.0
- Operating System: Windows
- Power Supply: USB

## Problem Description
Here's what happens when I try to run `idf.py menuconfig`

``` cmd
S:\esp\hello_world>idf.py menuconfig
Note: You are using Python 3.6.5. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from S:\esp\esp-idf\requirements.txt are satisfied.
Running cmake in directory S:\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized S:\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: S:/Program Files/Git/cmd/git.exe (found version "2.17.1.windows.2")
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at S:/esp/esp-idf/components/soc/CMakeLists.txt:4 (include):
  include could not find load file:

    /esp32/sources.cmake
Call Stack (most recent call first):
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  S:/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at S:/esp/esp-idf/CMakeLists.txt:39 (message):
  Failed to expand component requirements

-- Configuring incomplete, errors occurred!
See also "S:/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "S:/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

## Debug Logs

```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe 
[... 58 lines cut ...]

The output was:
1
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lsim
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhandlers-sim
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhal
collect2.exe: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: S:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe 
Build flags: -mlongcalls
Id flags:  

The output was:
1
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lsim
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhandlers-sim
s:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhal
collect2.exe: error: ld returned 1 exit status

```

projectgus, 2019-03-19:
Hi @souryavarenya ,

Do you possibly have a directory on your system named `S:\esp32\`?

This is a bug in the CMake support, which causes a check for this directory to pass incorrectly and then CMake tries to load files from it. Will fix, but in the meantime if you temporarily rename or move this directory then it should work around the issue.

souryavarenya, 2019-03-22:
Hey @projectgus,

Spot on! This fix works. 
Thanks!

---

This issue can be closed now!

projectgus, 2019-03-22:
Glad to hear that fixed it. We'll leave the issue open until the underlying CMake build system bug is fixed.

closed 2019-03-26 by igrr