espressif/esp-idf #3196: [BLE Mesh] Wrong BLE device name

ferremarco, 2019-03-18:
Adding 
```
esp_ble_mesh_set_unprovisioned_device_name("TEST");
```

in the initialization to change the device name, I see on the smartphone "TESTBLE-MESH"
I suppose is wrong BLE device sting length calculation when the advertisement packet is created.

```
static int ble_mesh_init(void)
{
    int err = 0;

    memcpy(dev_uuid + 2, esp_bt_dev_get_address(), ESP_BD_ADDR_LEN);

    esp_ble_mesh_register_prov_callback(esp_ble_mesh_prov_cb);
    esp_ble_mesh_register_custom_model_callback(esp_ble_mesh_model_cb);
    esp_ble_mesh_register_generic_client_callback(esp_ble_mesh_generic_cb);

    err = esp_ble_mesh_init(&provision, &composition);
    if (err) {
        ESP_LOGE(TAG, "Initializing mesh failed (err %d)", err);
        return err;
    }

    esp_ble_mesh_set_unprovisioned_device_name("TEST");
    esp_ble_mesh_node_prov_enable(ESP_BLE_MESH_PROV_ADV | ESP_BLE_MESH_PROV_GATT);

    ESP_LOGI(TAG, "BLE Mesh node initialized");

    board_led_operation(LED_G, LED_ON);

    return err;
}

```

Campou, 2019-03-18:
Thanks for reporting this issue, it has been fixed in v0.6 beta version which will be released soon.

zhanlej, 2019-10-25:
@Campou 

Why does the name scanned on the IOS always be "ESP32" after calling the ``esp_ble_mesh_set_unprovisioned_device_name()`` function to set a new name?

Campou, 2019-10-25:
Hi @zhanlej 

1. For Android, the name displayed on the App is the `local name` of the bluetooth device, which can be got from the **scan response data**.
2. For iOS, the name displayed on the App is the string contained in the **Device Name characteristic**. And this name can only be got after the App is connected with ESP32 successfully, and the App will read the characteristic value (i.e. name of the device). Then if the App disconnects with the device and starts to scan, the displayed name will be the correct one.

Currently because BLE Mesh registers a specific callback for GATT, so the above **ATT Read Request** will not be handled internally.

Thanks for reporting this, we will fix it.

Alvin1Zhang, 2019-10-28:
Thanks for reporting. Will reopen this until the fix is merged, thanks. @Campou @zhanlej

Alvin1Zhang, 2020-05-29:
Thanks for reporting and sorry for slow turnaround, the fix has been available at 093cc494b9db72a9e3e4ec76608e3388f192eccf, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-29 by Alvin1Zhang