espressif/esp-idf #3197: [BLE Mesh] Timeout status event

ferremarco, 2019-03-18:
Hello,

starting from your ["ESP BLE Mesh Client Model Demo"](https://github.com/espressif/esp-idf/tree/feature/esp-ble-mesh-v0.5/examples/bluetooth/ble_mesh/ble_mesh_client_model), I created a simple BLE Mesh Client Model (with a Generic OnOff Client Element) that send the "Generic OnOff Set message" to another node when receiving an event from a connected GPIO.

Here the function invoked:

```
int ble_mesh_update_client(int value)
{
  esp_ble_mesh_client_common_param_t common = {0};
  int err = 0;
  esp_ble_mesh_generic_client_set_state_t set_state = {0};
  common.opcode = ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET;
  common.model = onoff_client.model;
  common.ctx.net_idx = node_net_idx;
  common.ctx.app_idx = onoff_cli_pub.app_idx;
  common.ctx.addr = onoff_cli_pub.publish_addr;
  common.ctx.send_ttl = MSG_SEND_TTL;
  common.ctx.send_rel = MSG_SEND_REL;
  common.msg_timeout = MSG_TIMEOUT;
  common.msg_role = MSG_ROLE;
  set_state.onoff_set.op_en = false;
  set_state.onoff_set.onoff = value;
  set_state.onoff_set.tid = msg_tid;
  set_state.onoff_set.delay = 50;
  set_state.onoff_set.trans_time = 100;

  ESP_LOGI(TAG, "%s: TRY.... /n", __func__);
  err = esp_ble_mesh_generic_client_set_state(&common, &set_state);

  if (err != ESP_OK) {
    ESP_LOGE(TAG, "%s: Generic OnOff Set failed", __func__);
  }
  ESP_LOGI(TAG, "%s: OK !! /n", __func__);
  return err;
}
```

I've noticed that the ESP_BLE_MESH_GENERIC_CLIENT_TIMEOUT_EVT is often invoked (meny times) before receive the expected ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT.
That mean that on the Server device, I can see status change after (N * reveived ESP_BLE_MESH_GENERIC_CLIENT_TIMEOUT_EVT)
I've tested the Client with different Server implementation from different manufacturer. In attach the log.

Do you have suggestion in order to fix the issue ?

Regards,
Marco.
[BleClientModel.txt](/uploads/560102d04b3fd1c2343dd6efee3eed92/BleClientModel.txt)

Campou, 2019-03-18:
Hi @ferremarco ,
What is the value of the address "onoff_cli_pub.publish_addr"? And if it's convenient, could you please share the demo with us to have some test.
Thanks.

ferremarco, 2019-03-18:
Hi @Campou ,

in attach the example code.
[ble_mesh_client_example.zip](https://github.com/espressif/esp-idf/files/2977207/ble_mesh_client_example.zip)

ferremarco, 2019-03-19:
Hi @Campou ,

are there any news ?
Thanks,
Marco.

WCCWCC, 2019-03-20:
You are very likely that the address of another node is not correct.You should check onoff_cli_pub.publish_addr value. 'BleClientModel.txt' I can't download.

Alvin1Zhang, 2019-05-16:
@ferremarco Thanks for reporting the issue. Would you please share if any updates for the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2019-10-11:
@ferremarco Thanks for reporting, will close this ticket due to short of feedback, feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang