espressif/esp-idf #3200: assert in spi_device_transmit() line 1267

ammaree, 2019-03-18:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [n/a]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):  v4.0-dev-76-g96aa08a0f
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-ng-1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:         [external 5V]

## Problem Description
assert in  spi_device_transmit() line 1267

### Expected Behavior
assert should not happen

### Actual Behavior
assert happens

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs
abort() was called at PC 0x400db3e3 on core 0
0x400db3e3: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

ELF file SHA256: e5bedfdafd3d2817cc7c647af6ab8e01e536c9a6c453d0c046d2b796d06191e7

Backtrace: 0x4008974c:0x3ffcf7b0 0x40089a2d:0x3ffcf7d0 0x400db3e3:0x3ffcf7f0 0x40084f35:0x3ffcf80 0x400d9257:0x3ffcf850 0x400d9d7c:0x3ffcf8a0 0x400d9f88:0x3ffcf8c0 0x400d8717:0x3ffcf8e0 0x400d8828:0x3ffcf900
0x4008974c: invoke_abort at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/panic.c:715

0x40089a2d: abort at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/panic.c:715

0x400db3e3: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40084f35: spi_device_transmit at C:/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/spi_master.c:547

0x400d9257: m90e26Read at C:/Dropbox/devs/ws/z-components/devices/m90e26/m90e26.c:553

0x400d9d7c: m90e26GetSysStatus at C:/Dropbox/devs/ws/z-components/devices/m90e26/m90e26.c:506
 (inlined by) m90e26ReportSystem at C:/Dropbox/devs/ws/z-components/devices/m90e26/m90e26.c:613

0x400d9f88: m90e26Report at C:/Dropbox/devs/ws/z-components/devices/m90e26/m90e26.c:700 (discriminator 3)

0x400d8717: vCommandInterpret at C:/Dropbox/devs/ws/z-appl/iMacs/task_control.c:322

0x400d8828: vTaskControl at C:/Dropbox/devs/ws/z-appl/iMacs/task_control.c:430

Rebooting...

## Other items if possible

I believe the crash occurs due to an SPI transaction already being in progress since we do have a higher priority task running in the background that is reading the sensor.

The todo note in line 1260 is a hint that this checking and locking still needs to be done?

ammaree, 2019-03-19:
I have added a mutex to protect the SPI driver and the problem has not yet occurred again. I think we can accept that this proves the statement in the todo note and that the driver should be updated.

ginkgm, 2019-03-28:
Hi @ammaree ,

When you see the assertion, it means you failed to handle the return value of previous transaction (by ``spi_device_get_trans_result``) while you are trying to get the return of a new one. If this assertion is removed, you will incorrectly treat the returning value of previous transation as what you just get.

SPI is usually used in a sequence-sensitive way, which means, when you are talking to a slave, you have to follow a special sequence. Any violation to the access sequence (determined by the device) may cause unexpected issue.

When communicating with a device (slave), you should always deal with the transaction (send-wait-read) one by one. The driver offers a sending queue and a return queue to ensure that every transaction you send first will return its result first. It's not allowd (at least currently not) for you to skip the read value handling of previous transaction, but get the reading of latest transaction first. If you can provide a spec of your slave device, maybe we can provide the feature (e.g. allow two devices sharing a CS pin) in the future.

The ``device`` conception in the driver requires you to maintain the send/receive sequence to the device, no matter how many tasks you use. Basicly, we suggest our user to, if they don't want to know how the driver work, ALWAYS communicate with a device in ONLY ONE task. If you only use spi_device_transmit, the case is even simpler, you don't even have to care about how many transaction you have already sent.

ginkgm, 2019-03-28:
BTW, your solution is quite good, as we usually do. Use a mutex to protect a series of transactions (whose sequence should be ensured) from other (series of transaction) is a great idea.

If you stick to not using Mutex, there is another way. Add two devices without CS lines. Toggles the GPIO pins in the pre_trans_cb and post_trans_cb for both devices. This can let you maintain transaction queue on device 2 while able to access to device 1 at the same time. But please take your own risk for breaking the transaction sequence of the slave.

ammaree, 2019-05-03:
Closed, fixed by #3394 until permanent fix has come through

closed 2019-05-03 by ammaree