espressif/esp-idf #3201: Crash occuring during TLS reset

michelesolytic, 2019-03-19:
## Environment

- Development Kit: custom board
- Module or chip used: ESP32-WROOM-32D
- IDF version: v3.1.3
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux

## Problem Description
Occasionally the reset of a TLS session crashes with a guru meditation error. Problem also occurs with sdkconfig reset at default values.

## Debug Logs
D (1552925543.728727) HTTP_CLIENT: Code: 2999; Message: D (318819) HTTP_CLIENT: Begin connect to: https://xxxxxxxxxxx:443[0m
D (1552925543.741048) TRANS_SSL: Code: 2999; Message: D (318829) TRANS_SSL: Cleanup mbedtls[0m
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060930  A0      : 0x8017a1c4  A1      : 0x3ffd95f0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000414d  A5      : 0x00000000  
A6      : 0x3f422114  A7      : 0x00000414  A8      : 0x00000000  A9      : 0x3ffd9560  
A10     : 0x3ffe247c  A11     : 0x40082e2c  A12     : 0x3f422114  A13     : 0x00000bb7  
[0;33m0x40082e2c: _free_r at /home/solytic/git/esp-idf/components/newlib/syscalls.c:41
[0m
A14     : 0x3ffe247c  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000413  
Backtrace: 0x4000c46c:0x3ffd95f0 0x4017a1c1:0x3ffd9600 0x4017a225:0x3ffd9620 0x40176397:0x3ffd9640 0x401767c3:0x3ffd9660 0x4019a42d:0x3ffd96b0 0x40175bc2:0x3ffd96d0 0x40175da8:0x3ffd9710 0x4011fee1:0x3ffd9730 0x401227cc:0x3ffd9750 0x40122965:0x3ffd9940
[0;33m0x4017a1c1: ssl_session_reset_int at /home/solytic/git/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:6781
[0m
[0;33m0x4017a225: mbedtls_ssl_session_reset at /home/solytic/git/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:6781
[0m
[0;33m0x40176397: ssl_close at /home/solytic/git/esp-idf/components/esp_http_client/lib/transport_ssl.c:227 (discriminator 6)
[0m
[0;33m0x401767c3: ssl_connect at /home/solytic/git/esp-idf/components/esp_http_client/lib/transport_ssl.c:164
[0m
[0;33m0x4019a42d: transport_connect at /home/solytic/git/esp-idf/components/esp_http_client/lib/transport.c:223
[0m
[0;33m0x40175bc2: esp_http_client_open at /home/solytic/git/esp-idf/components/esp_http_client/esp_http_client.c:992
[0m
[0;33m0x40175da8: esp_http_client_perform at /home/solytic/git/esp-idf/components/esp_http_client/esp_http_client.c:992

mahavirj, 2019-03-25:
@michelesolytic Can you please also analyze heap statistics (minimum free heap, largest available block etc.) when this happens? Also if possible it would be great if you can share your application that successfully reproduces this problem.

michelesolytic, 2019-03-29:
Hello Mahavirj!
I don't have heap statistics with me, I will try to make some in the future. I am narrowing down the problem and I believe this occurs either when reusing a connection whose esp_http_client_perform() failed, and/or concurrent connections are taking place. I'll try to develop a test application to narrow down the issue.
Just to be sure: isn't the HTTP stack supposed to be able to handle concurrent connections?

mahavirj, 2019-05-06:
@michelesolytic 

>  Just to be sure: isn't the HTTP stack supposed to be able to handle concurrent connections?

Sorry I missed your question, yes this should work. Do you have any update on example code that can reproduce this issue reliably?

mahavirj, 2019-05-13:
@michelesolytic I will close this due to lack of response, but please feel free to raise issue if still exists.

closed 2019-05-13 by mahavirj