espressif/esp-idf #3202: a function name for closing socket is too universal.

keesus, 2019-03-19:
I recently encountered a frustrating situation.

At components/tcp_transport.c
 
There's a declared function name 'close' to close socket.
> if (connect(tcp->sock, (struct sockaddr *)(&remote_ip), sizeof(struct sockaddr)) != 0) {
    **close(tcp->sock);**
    tcp->sock = -1;
    return -1;
 }

What I struggled is socket was not closing so I couldn't connect multiple HTTP connections.

I was using another function named 'close' in my code and it caused a overlap with ESP-IDF declared function.

So I suggest changing this function name to 'close_socket' or something more specific.

negativekelvin, 2019-03-19:
Sure, just go back in time before Berkeley sockets was created in 1983 and establish your use of close function as the universal standard. https://en.m.wikipedia.org/wiki/Berkeley_sockets#Socket_API_functions
Or just rename **your** function.

AadiMehta, 2019-03-19:
@negativekelvin okay that was hillarious :D

keesus, 2019-03-20:
@negativekelvin 

Okay I was fool enough.
But I thought the actual function name could be different to original concept.
I admire your account name.

closed 2019-03-20 by keesus