espressif/esp-idf #3203: [BLE Mesh] Device identification during provisioning

ferremarco, 2019-03-19:
Is there a callback event related device identification procedure before provisioning ?

Thanks,
Marco.

Campou, 2019-03-25:
Hi @ferremarco,
1. Thanks for the question. Currently we don't implement any callback for the identification procedure.
2. In my opinion, the callback of identification is useful for BLE Mesh Provisioner, in which the Provisioner can choose the capbility of the invited unprovisioned device. But for BLE Mesh Node, it may be not that important.
3. And if you have any suggestions, please feel free to share with us.

Thanks.

chegewara, 2019-03-25:
Maybe i dont understand purpose of this callback, but you have OOB for this:
https://github.com/espressif/esp-idf/blob/feature/esp-ble-mesh-v0.5/examples/bluetooth/ble_mesh/ble_mesh_node/main/ble_mesh_demo_main.c#L87-L98

In example with LEDs you will have blinking LED for example, to confirm you are provisioning device you want.

WCCWCC, 2019-03-26:
Hi，
     A Provisioner  provisioning unprovisioned device .There are 4 ways to authenticate.

               Authentication with Output OOB
               Authentication with Input OOB
               Authentication with static OOB  
               Authentication with  no OOB  

    if Authentication failed,The unprovisioned device will not enter the mesh network.

Alvin1Zhang, 2019-05-16:
@ferremarco Thanks for reporting the issue. Would you help share if any updates for the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2019-10-11:
@ferremarco Thanks for reporting, will close this ticket due to short of feedback, feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang