espressif/esp-idf #3204: New errror: wifi noTIM!!

ammaree, 2019-03-19:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [n/a]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): latest as of today 2019/3/19
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): v5.2.0
- Operating System:     [Windows]
- Power Supply:         [external 3.3V]

## Problem Description
At random times and without and identifiable set of preceding events the error message appears. Have searched the web for the error message but very little available other than an explanation that it normally indicates the absence of Traffic Indication Map info.
[https://en.wikipedia.org/wiki/Traffic_indication_map](url)

### Expected Behavior
No error message and no loss of communications

### Actual Behavior
Get error message and sometimes loss of communications requiring a reboot

### Steps to reproduce
Cannot reproduce at all.

### Code to reproduce this issue

## Debug Logs

## Other items if possible

ammaree, 2019-03-19:
The noTIM!! error message is often followed by another unknown new/error being:
wifi wifi state wrong txa_flags=9

Any chance of shedding some light on these failures?

liuzfesp, 2019-03-21:
HI @ammaree , "wifi wifi state wrong txa_flags=9" doesn't indicate a bug, it means ESP32 has sent the AddBA request to AP but AP doesn't reply AddBA response within specified time (1 second). Maybe we should remove the log in future.

I think "sometimes loss of communications requiring a reboot" should have no relationship with the log, it also has no relationship with TIM.

Need more detailed description about the problem before we can investigate it, e.g.:

- Does "sometimes loss of communications requiring a reboot" mean WiFi disconnect or socket disconnect?
- log

ammaree, 2019-05-03:
I think we can close this issue. The errors have gone away, not sure whether as a result of the AP reconfiguration or the latest Wifi libraries.

closed 2019-05-03 by ammaree