espressif/esp-idf #3205: [Wifi APSTA] Fail to connect (STA) while AP on - v3.2-beta3

eftimarco, 2019-03-19:
Hello,

Upgrading from 3.1.2-114 to v3.2-beta3 to get http_server, I face a problem I did not get previously.
- Wifi mode is APSTA, AP is working, a telephone is connected to the esp-32.
- I have to test a wifi (ssid - password). This is what I run : 
	wifi_config_t wifi_config_sta;
	strncpy((char *) wifi_config_sta.sta.ssid, ssid_p, WIFI_SSID_MAX);
	strncpy((char *) wifi_config_sta.sta.password, password_p, WIFI_PWD_MAX);
	esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config_sta);
	esp_wifi_connect();
I never get neither SYSTEM_EVENT_STA_CONNECTED nor SYSTEM_EVENT_STA_GOT_IP...

The same code was working well on esp-idf 3.1.2-114.

Thanks in advance

eftimarco, 2019-03-20:
More details on the problem faced :
- Every call get a ESP_OK return 
- I get a SYSTEM_EVENT_STA_DISCONNECTED on the connect, with reason : WIFI_REASON_NO_AP_FOUND
- If I do esp_wifi_stop(), esp_wifi_set_mode(WIFI_MODE_STA), esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) and esp_wifi_start() instead, to test the same wifi (ssid - password) at the same place in code, it is connecting, but telephone connected to esp-32 in AP mode is lost, obviously

liuzfesp, 2019-03-21:
Hi @eftimarco does this problem exist if wifi_config_sta.sta.scan_method is configured to WIFI_ALL_CHANNEL_SCAN.

eftimarco, 2019-03-21:
Hi @liuzfesp, 

Same problem (NO_AP_FOUND) with wifi_config_sta.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;

- Other test : Wifi scanning is working well during APSTA mode, when the telephone is connected to esp-32, with that code : 
wifi_scan_config_t scan_config = {
	.ssid = NULL,
	.bssid = NULL,
	.channel = 0,
	.show_hidden = false,
	.scan_type = WIFI_SCAN_TYPE_ACTIVE
};
err = esp_wifi_scan_start(&scan_config, true);
[...]
esp_wifi_scan_get_ap_num(&nb_ap_scaned);
[...]
esp_wifi_scan_get_ap_records(&nb_ap_scaned, wifi_ap_founded);

liuzfesp, 2019-03-21:
Hi @eftimarco just can't reproduce this issue. Could you let me know the AP type to which the ESP32 is connected?

eftimarco, 2019-03-21:
@liuzfesp Both the 2 I try to connect to are 802.11n with WPA2-PSK encryption

liuzfesp, 2019-03-21:
Hi @eftimarco, what I want to know is the vendor name of the AP, for example, the API is TP-LINK TL-WR1043 etc.

eftimarco, 2019-03-21:
The main one is a tp-link TL-WR802N 
(Second one is an internet box).

liuzfesp, 2019-03-21:
I just did the test on my TPLINK TL-WR845N, everything looks good. Is it possible for you to write a demo code to produce this issue and paste the demo code here. You can write the demo based on IDF WiFi example, such as $IDF_PATH/examples/wifi/getting_started/station.

eftimarco, 2019-03-21:
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via 'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

#define EXAMPLE_ESP_WIFI_AP_SSID   "MY_WIFI"
#define EXAMPLE_ESP_WIFI_AP_PASS   "MY_WIFI_PASS"

#define WIFI_SSID_MAX        32
#define WIFI_PWD_MAX         64

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

static system_event_id_t wifi_sta_state = SYSTEM_EVENT_STA_STOP;
static system_event_id_t wifi_ap_state = SYSTEM_EVENT_AP_STOP;

/* The event group allows multiple bits for each event, but we only care about one event 
 * - are we connected to the AP with an IP? */
#define WIFI_CONNECTED_BIT          BIT0
#define WIFI_STOP_BIT               BIT1
#define WIFI_DISCONNECT_DONE_BIT    BIT2
#define WIFI_SCAN_DONE_BIT          BIT3
#define WIFI_PHONE_CONNECTED        BIT4
#define WIFI_AP_STARTED             BIT5

static const char *TAG = "wifi test";

/*
    Associe des actions a chaque changement d'etat du WiFi
*/
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id){

        case SYSTEM_EVENT_WIFI_READY : // Deprecated
        break;

        case SYSTEM_EVENT_SCAN_DONE:
            xEventGroupSetBits(s_wifi_event_group, WIFI_SCAN_DONE_BIT);
            break;

        // STA Part
        case SYSTEM_EVENT_STA_START:
            xEventGroupClearBits(s_wifi_event_group, WIFI_STOP_BIT);
[... 263 lines cut ...]
{
    EventBits_t uxBits;
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    WIFI_event_init();
    // Connecting there to the AP EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS in STA MODE works
    //wifi_init_sta(); 
    if(WIFI_ap_init()){
        ESP_LOGI(TAG, "AP INIT Ok - SSID : %s - PASSWORD : %s", EXAMPLE_ESP_WIFI_AP_SSID, EXAMPLE_ESP_WIFI_AP_PASS);
    }
    uxBits = xEventGroupWaitBits(s_wifi_event_group, SYSTEM_EVENT_AP_STACONNECTED, pdTRUE, pdFALSE, (10000 / portTICK_PERIOD_MS) );
    if(uxBits & SYSTEM_EVENT_AP_STACONNECTED){
        ESP_LOGI(TAG, "A device is connected to us (AP mode)");
    }
    else{
        ESP_LOGI(TAG, "We waited 10 sec but no station connect to us (AP mode) - trying connection (STA mode) anyway");
    }
    WIFI_test_ssid(EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
}

eftimarco, 2019-03-21:
Here it is :
[station_example_main.txt](https://github.com/espressif/esp-idf/files/2992866/station_example_main.txt)

liuzfesp, 2019-03-22:
HI @eftimarco， the root cause is because you forget to call memset(&wifi_config_sta, 0, sizeof(wifi_config_sta)) to initialize wifi_config_sta.

BTW, another typepo in app_main() is to wait SYSTEM_EVENT_AP_STACONNECTED, it should be WIFI_PHONE_CONNECTED, anyway I don't think it's a good idea to block the main task before some phones are connected to the ESP32 AP.

eftimarco, 2019-03-22:
Hello @liuzfesp, 
I tried with the memset : 
```
    memset(&wifi_config_sta, 0, sizeof(wifi_config_sta));
    strncpy((char *) wifi_config_sta.sta.ssid, ssid_p, WIFI_SSID_MAX);
    strncpy((char *) wifi_config_sta.sta.password, password_p, WIFI_PWD_MAX);
    wifi_config_sta.sta.scan_method = WIFI_ALL_CHANNEL_SCAN;
    if(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config_sta) == ESP_OK){
        if(esp_wifi_connect() == ESP_OK)
```

**I still have the same issue** :
```
I (11473) wifi test: Test du wifi : SSID = 'xxx' ; password = 'xxx'
I (14313) wifi test: SYSTEM_EVENT_STA_DISCONNECTED - Disconnection reason : 201
I (14313) wifi test: Event 0x5, wifi_sta_state : 0x5 - wifi_ap_state = 0xD
E (23483) wifi test: Impossible to connect the wifi... (password ?)
```

Right for the mistake SYSTEM_EVENT_AP_STACONNECTED => WIFI_PHONE_CONNECTED !
The blocking was for the example, waiting for the phone to connect, but you're right : we have the same issue if no phone is connected (but the AP configured).
Thank you for your help

liuzfesp, 2019-03-22:
HI @eftimarco， everything looks good on my side. Please add retry on receiving SYSTEM_EVENT_STA_DISCONNECTED:
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
...
        case SYSTEM_EVENT_STA_DISCONNECTED: //
            xEventGroupClearBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
            xEventGroupSetBits(s_wifi_event_group, WIFI_DISCONNECT_DONE_BIT);
            wifi_sta_state = event->event_id;
            ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED - Disconnection reason : %d\n", event->event_info.disconnected.reason);
            **_//try again...
            esp_wifi_connect();_**
            break;
...
}

eftimarco, 2019-03-22:
@liuzfesp First test does not work : 
```
I (14336) wifi test: SYSTEM_EVENT_STA_DISCONNECTED - Disconnection reason : 201
I (14336) wifi test: Event 0x5, wifi_sta_state : 0x5 - wifi_ap_state = 0xD
I (17176) wifi test: SYSTEM_EVENT_STA_DISCONNECTED - Disconnection reason : 201
I (17176) wifi test: Event 0x5, wifi_sta_state : 0x5 - wifi_ap_state = 0xD
I (20016) wifi test: SYSTEM_EVENT_STA_DISCONNECTED - Disconnection reason : 201
I (20016) wifi test: Event 0x5, wifi_sta_state : 0x5 - wifi_ap_state = 0xD
```
I tried unsetting Store phy calibration data in NVS, recompiling and running... same issue...

liuzfesp, 2019-03-22:
@eftimarco， so it's very strange. Could you double check whether the AP that ESP32 STA tries to connect is correctly configured the SSID and password? Does the AP set to open mode?

eftimarco, 2019-03-22:
You're right ! The solution was :
```
memset(&wifi_config_sta, 0, sizeof(wifi_config_sta));
```
I had a problem on wifi hub today (that I did not have yesterday) : that solved, it works !
Thanks you @liuzfesp !!!

closed 2019-03-22 by eftimarco