espressif/esp-idf #3206: [FreeRTOS QUEUE] Possible bug introduced by commit #13523c9 (Dec 20, 2018)

mws-rmain, 2019-03-19:
I believe the logic below introduced by git commit #13523c9 (Dec. 20, 2019, 'freertos: check that mutex is released by owner task') is faulty.

In FreeRTOS queue.c:727
```
#if ( configUSE_MUTEXES == 1 && configCHECK_MUTEX_GIVEN_BY_OWNER == 1)
{
	configASSERT(
              pxQueue->uxQueueType != queueQUEUE_IS_MUTEX 
              || pxQueue->pxMutexHolder == NULL 
              || xTaskGetCurrentTaskHandle() == pxQueue->pxMutexHolder
         );
}
#endif
```

configASSERT() triggers if the logic tests FALSE, so I believe the intended logic should be:

```
#if ( configUSE_MUTEXES == 1 && configCHECK_MUTEX_GIVEN_BY_OWNER == 1)
{
	configASSERT(~(
              (pxQueue->uxQueueType == queueQUEUE_IS_MUTEX) 
              && (pxQueue->pxMutexHolder != NULL) 
              && (xTaskGetCurrentTaskHandle() != pxQueue->pxMutexHolder)
        ));
}
#endif
```
(i.e., if (this is a MUTEX-type queue AND there is a valid MUTEX holder AND the current task is NOT the mutex holder), then trigger the ASSERT).

Please confirm?

igrr, 2019-03-20:
I think the two snippets you have posted should be equivalent, as  !(!a && !b) == a || b. Do you see a specific set of values where one does the right thing but the other does not?

mws-rmain, 2019-03-20:
Thanks, Ivan.
Looking at it again, I don't know how I convinced myself this was the cause of my problem.
Thanks.

closed 2019-03-20 by mws-rmain