espressif/esp-idf #3207: nvs_set_blob() then nvs_commit() then nvs_get_blob() returns NVS_NOT_FOUND

jeremyherbert, 2019-03-20:
## Environment

- Module or chip used:  ESP32-WROVER (bare module)
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-91-gb7b5c3a8c
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         External 3.3V

## Problem Description

When writing a blob using NVS, committing and then reading the blob back, an error occurs which indicates that the key is missing.

### Expected Behavior

Everything should work

### Actual Behavior

Fails at the line marked with `// fails here`

### Code to reproduce this issue

main.c: 
```cpp
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "string.h"

const char *TAG = "MAIN";

static inline void check_retval(esp_err_t resp) {
    if (resp != ESP_OK) {
        ESP_LOGE(TAG, "NVS error: %s", esp_err_to_name(resp));
        abort();
    }
}

void app_main()
{
    const char *partition_name = "nvs";
    const char *namespace = "test";
    const char *key = "abc";
    const char *str_val = "xyz";

    nvs_handle handle;

    esp_err_t ret = nvs_flash_init_partition(partition_name);

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_LOGI(TAG, "Erase required for partition %s", partition_name);
        check_retval(nvs_flash_erase());
        check_retval(nvs_flash_init_partition(partition_name));
    } else {
        check_retval(ret);
    }
    ESP_LOGI(TAG, "Opening partition %s...", partition_name);
    check_retval(nvs_open_from_partition(partition_name, namespace, NVS_READWRITE, &handle));
    ESP_LOGI(TAG, "Opening partition %s succeeded", partition_name);

    uint8_t value;
    if(nvs_get_u8(handle, key, &value) == ESP_ERR_NVS_NOT_FOUND) {
        ESP_LOGI(TAG, "u8 not found");
    }

    value = 23;
    check_retval(nvs_set_u8(handle, key, value));
    check_retval(nvs_commit(handle));

    value = 0;
    check_retval(nvs_get_u8(handle, key, &value));
    configASSERT(value == 23);
    ESP_LOGI(TAG, "u8 ok!");

    check_retval(nvs_set_blob(handle, key, str_val, 3));
    check_retval(nvs_commit(handle));
    ESP_LOGI(TAG, "blob set");

    char tmp[3];
    size_t len;
    check_retval(nvs_get_blob(handle, key, tmp, &len)); // fails here, ESP_ERR_NVS_NOT_FOUND
    ESP_LOGI(TAG, "blob read");
    configASSERT(len == 3);
    configASSERT(memcmp(str_val, key, len) == 0);
    ESP_LOGI(TAG, "blob ok!");

    while(1) {
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}
```

partitions.csv:
```
# Name,        Type, SubType, Offset,  Size
nvs,           data, nvs,     0x9000,  0x4000
otadata,  

	data, ota,     0xd000,  0x2000
phy_init, 	data, phy,     0xf000,  0x1000
factory,  	0,    factory, 0x10000, 1M
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2986240/sdkconfig.txt)

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6328
load:0x40078000,len:12360
load:0x40080400,len:7136
entry 0x40080760
I (28) boot: ESP-IDF v4.0-dev-91-gb7b5c3a8c 2nd stage bootloader
I (29) boot: compile time 12:06:46
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) boot: Defaulting to factory image
I (96) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x137b8 ( 79800) map
I (133) esp_image: segment 1: paddr=0x000237e0 vaddr=0x3ffb0000 size=0x01f3c (  7996) load
I (137) esp_image: segment 2: paddr=0x00025724 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/jeremy/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (140) esp_image: segment 3: paddr=0x00025b2c vaddr=0x40080400 size=0x0a4e4 ( 42212) load
I (167) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x1d2d0 (119504) map
0x400d0018: _flash_cache_start at ??:?

I (209) esp_image: segment 5: paddr=0x0004d2f0 vaddr=0x4008a8e4 size=0x03c6c ( 15468) load
0x4008a8e4: prvProcessTimerOrBlockTask at /home/jeremy/esp/esp-idf/components/freertos/timers.c:571

I (225) boot: Loaded app from partition at offset 0x10000
I (225) boot: Disabling RNG early entropy source...
I (227) spiram: Found 64MBit SPI RAM device
I (230) spiram: SPI RAM mode: flash 40m sram 40m
I (235) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (242) cpu_start: Pro cpu up.
I (246) cpu_start: Application information:
I (251) cpu_start: Project name:     hello-world
I (256) cpu_start: App version:      1
I (261) cpu_start: Compile time:     Mar 20 2019 12:06:40
I (267) cpu_start: ELF file SHA256:  b430c3b904f00b08...
I (273) cpu_start: ESP-IDF:          v4.0-dev-91-gb7b5c3a8c
I (279) cpu_start: Starting app cpu, entry point is 0x40081238
0x40081238: call_start_cpu1 at /home/jeremy/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (1160) spiram: SPI SRAM memory test OK
I (1160) heap_init: Initializing. RAM available for dynamic allocation:
I (1161) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1167) heap_init: At 3FFB3640 len 0002C9C0 (178 KiB): DRAM
I (1173) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1179) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1186) heap_init: At 4008E550 len 00011AB0 (70 KiB): IRAM
I (1192) cpu_start: Pro cpu start user code
I (1197) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (548) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (550) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (580) MAIN: Opening partition nvs...
I (580) MAIN: Opening partition nvs succeeded
I (580) MAIN: u8 not found
I (580) MAIN: u8 ok!
I (590) MAIN: blob set
E (590) MAIN: NVS error: ESP_ERR_NVS_NOT_FOUND
abort() was called at PC 0x400d56d8 on core 0
0x400d56d8: check_retval at /home/jeremy/esp_test/build/../main/hello_world_main.c:22 (discriminator 2)

ELF file SHA256: b430c3b904f00b0831f5e26547de9bf9067aed503d7800e6ce3a54223694408c

Backtrace: 0x40087784:0x3ffb5300 0x400879e9:0x3ffb5320 0x400d56d8:0x3ffb5340 0x400d583f:0x3ffb5360 0x400d0dd0:0x3ffb5390 0x40087c59:0x3ffb53b0
0x40087784: invoke_abort at /home/jeremy/esp/esp-idf/components/esp32/panic.c:715

0x400879e9: abort at /home/jeremy/esp/esp-idf/components/esp32/panic.c:715

0x400d56d8: check_retval at /home/jeremy/esp_test/build/../main/hello_world_main.c:22 (discriminator 2)

0x400d583f: app_main at /home/jeremy/esp_test/build/../main/hello_world_main.c:68 (discriminator 9)

0x400d0dd0: main_task at /home/jeremy/esp/esp-idf/components/esp32/cpu_start.c:530

0x40087c59: vPortTaskWrapper at /home/jeremy/esp/esp-idf/components/freertos/port.c:435
```

igrr, 2019-03-20:
Thanks for reporting this. Are you intentionally using the same key name for u8 and then for blob? Do you observe the same issue if the blob is stored under a different key name?

jeremyherbert, 2019-03-20:
Sorry, that was a bad copy paste; I was extracting this code from a much larger piece of code that I can't share

This also causes the same problem with same backtrace (I did an `erase_flash` before reflashing this):
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "string.h"

const char *TAG = "MAIN";

static inline void check_retval(esp_err_t resp) {
    if (resp != ESP_OK) {
        ESP_LOGE(TAG, "NVS error: %s", esp_err_to_name(resp));
        abort();
    }
}

void app_main()
{
    const char *partition_name = "nvs";
    const char *namespace = "test";
    const char *key = "abc";
    const char *key2 = "abc2";
    const char *str_val = "xyz";

    nvs_handle handle;

    esp_err_t ret = nvs_flash_init_partition(partition_name);

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_LOGI(TAG, "Erase required for partition %s", partition_name);
        check_retval(nvs_flash_erase());
        check_retval(nvs_flash_init_partition(partition_name));
    } else {
        check_retval(ret);
    }
    ESP_LOGI(TAG, "Opening partition %s...", partition_name);
    check_retval(nvs_open_from_partition(partition_name, namespace, NVS_READWRITE, &handle));
    ESP_LOGI(TAG, "Opening partition %s succeeded", partition_name);

    uint8_t value;
    if(nvs_get_u8(handle, key, &value) == ESP_ERR_NVS_NOT_FOUND) {
        ESP_LOGI(TAG, "u8 not found");
    }

    value = 23;
    check_retval(nvs_set_u8(handle, key, value));
    check_retval(nvs_commit(handle));

    value = 0;
    check_retval(nvs_get_u8(handle, key, &value));
    configASSERT(value == 23);
    ESP_LOGI(TAG, "u8 ok!");

    check_retval(nvs_set_blob(handle, key2, str_val, 3));
    check_retval(nvs_commit(handle));
    ESP_LOGI(TAG, "blob set");

    char tmp[3];
    size_t len;
    check_retval(nvs_get_blob(handle, key2, tmp, &len)); // fails here, ESP_ERR_NVS_NOT_FOUND
    ESP_LOGI(TAG, "blob read");
    configASSERT(len == 3);
    configASSERT(memcmp(tmp, str_val, len) == 0);
    ESP_LOGI(TAG, "blob ok!");

    while(1) {
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}
```

jeremyherbert, 2019-03-20:
I think I have found the problem here (well, two problems, one is my fault). There isn't an API that can be used when you already know that the buffer is large enough to read the blob; in the case of this code, `len` was not being initialised to the length of the buffer. The main issue here is that the wrong error was returned; if you do this same thing with `nvs_get_str`, it returns `ESP_ERR_NVS_INVALID_LENGTH` which is correct.

jeremyherbert, 2019-03-20:
Can I also suggest adding a new NVS error type, which is something along the lines of NVS_INVALID_TYPE which is basically that the key exists, but the type is wrong. For example, if you set a bool but get as a u8, it should return this error.

sagb2015, 2019-03-28:
Hi @jeremyherbert , I tried your example and it seems to return expected value which is ESP_ERR_NVS_INVALID_LENGTH when length is uninitialised/less than 3. The blob is read fine for len=3. I was unable to reproduce ESP_ERR_NVS_NOT_FOUND issue. It will help if you can mention the IDF commit id and also attach sdkconfig.

jeremyherbert, 2019-03-29:
Hi @sagb2015 

v4.0-dev-91-gb7b5c3a8c

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3021349/sdkconfig.txt)

sagb2015, 2019-04-18:
@jeremyherbert - I could not reproduce the issue @v4.0-dev-91-gb7b5c3a8c
PS - Apologise for asking for the details that you had already provided.

jeremyherbert, 2019-04-18:
Ok, let’s close this and I will reopen if it comes up again.

closed 2019-04-18 by jeremyherbert