espressif/esp-idf #3208: Seperate return type and function name with space in CAN document

keelung-yang, 2019-03-20:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/can.html#api-reference

[image]

and so on...

keelung-yang, 2019-03-27:
The same as in SD document:
[image]
If I want to copy function name by double click, I got return type and connected funcion name.

IT SO WEIRD why are you always ignore this? 
I'm confused that you tread this such long long time.

Dazza0, 2019-08-01:
Hi @keelung-yang , thanks for reporting this issue.

I think this may be a HTML/browser rendering issue, as I can only recreate this when using Chrome. A little bit of background, we use Sphinx to auto-generate our web documentation, and Sphinx seems to put non-standard C types such as `esp_err_t`  and function names within HTML `<span>` tags. It seems like Chrome is not does not render any white-space between `<span>` tags.

We'll look into fixing this with a CSS override, and if that doesn't work we can raise this issue with the Sphinx maintainers.

closed 2019-08-21 by igrr