espressif/esp-idf #3210: esp_wifi_80211_tx transmits only on channel 4.

ingframin, 2019-03-20:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|]

## Problem Description

No matter which function I call to change channel, esp_wifi_80211_tx transmits only on channel 4. 
I tried:
1) esp_wifi_set_config(&conf) where cone had both confidence.ap.channel and con.sta.channel set to another channel
2) esp_wifi_set_channel

3) In Arduino, WiFi.softAP("none","",channel,0,5)

Is this expected behaviour or is there something wrong with my devices?

### Expected Behavior
I should be able to transmit a beacon on whatever channel I select.

### Actual Behavior
It only transmits beacons on channel 4

### Steps to repropduce

I can provide my Arduino source code if needed.

liuzfesp, 2019-03-21:
HI @ingframin does the STA connect to a router? If yes, what's the channel of the router?

imbearr, 2019-04-04:
@liuzfesp is absolutely right! ESP modules have **software** acces point, so if you module connected to router it must have same channel for AP becaouse phisycal layer can't work on to different channels simultaneously.

Alvin1Zhang, 2019-05-20:
@ingframin Hi, thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-05-22:
@ingframin Thanks for reporting the issue, feel free to reopen the issue if it still exists, thanks.

closed 2019-05-22 by Alvin1Zhang