espressif/esp-idf #3211: ESP32 code can be compiled with global static arrays that cause FreeRTOS crash at runtime.

marcmerlin, 2019-03-20:
As referenced by @projectgus in https://github.com/espressif/arduino-esp32/issues/2567
"the (apparently new) case of arrays fitting at compile time but taking enough space that FreeRTOS crashes at startup? (assert on malloc fail as you pointed out)."

Compile output:
Global variables use 110860 bytes (33%) of dynamic memory, leaving 216820 bytes for local variables. Maximum is 327680 bytes.

crash before changing to malloc:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
0x40088960: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x40088b63: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x40082b72: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 381
0x40082d04: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 213

If the array is too big, the compiler catches that and says it won't fit.
If it's a bit smaller, the compiler/linker succeeds, but freertos crashes at runtime before arduino setup() can be run.

marcmerlin, 2019-04-05:
@projectgus  did you have someone you wanted to assign this to?

projectgus, 2019-04-08:
Hi @marcmerlin ,

The internal ticket that tracks this one is assigned to me. Will update here once we have any progress.

Angus