espressif/esp-idf #3212: Secure boot (Reflashable) issue - Continuously resetting

JoaoBerlese, 2019-03-20:
### Environment

- Development Kit: ESP32-DevKitC (v2)
- IDF version, I tried in more than one version:
    v4.0-dev-82-gf19ecbbbf
    v3.2-beta1-15-g12b2268ee
    v2.5.0 ([link forum](https://www.esp32.com/viewtopic.php?t=6579))
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

### Problem Description

Secure boot doesn't work, despite all my attempts, always end up in a error resetting looping as show below:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8532
load:0x40078000,len:17148
load:0x40080400,len:6680
csum err:0xa2!=0xff
Falling back to built-in command interpreter.
OK
```

### Steps to repropduce
I followed the step by step instructions described in [Re-Flashable Software Bootloader](https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#re-flashable-software-bootloader).

0. make menuconfig (command)
   Security features-->
   [image]

1. Generate signing key:
   `python $IDF_PATH/components/esptool_py/esptool/espsecure.py generate_signing_key secure_boot_signing_key.pem`

2. make bootloader (command)

3. Write Efuse block 2
   `python /home/user/esp/esp-idf/components/esptool_py/esptool/espefuse.py burn_key secure_boot /home/user/esp/projects/blink/build/bootloader/secure-bootloader-key-256.bin`

4. Write bootloader
   `python /home/user/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/user/esp/projects/blink/build/bootloader/bootloader.bin`

4.1 Reflash bootloader
   `python /home/user/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 /home/user/esp/projects/blink/build/bootloader/bootloader-reflash-digest.bin`

5. make flash monitor (command)
   And is here where I see the looping error being printed out.
---

I even tried this other command...

Enabling Secure Boot mechanism:
`python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyUSB0 burn_efuse ABS_DONE_0`

But in the end I just blocked for ever one of my ESP32 kits. =[

bvernoux, 2019-03-20:
It seems this issue is fixed in forum:
by dmitryga » Wed Aug 01, 2018 3:34 pm 
> I've updated esptool.py to 2.5.0 and now everything works like a charm!
> Bootloader is booting, application is starting, flash is encrypted.
> ESP_Angus, thank you for your help!

JoaoBerlese, 2019-03-20:
> It seems this issue is fixed in forum:
> by dmitryga » Wed Aug 01, 2018 3:34 pm
> 
> > I've updated esptool.py to 2.5.0 and now everything works like a charm!
> > Bootloader is booting, application is starting, flash is encrypted.
> > ESP_Angus, thank you for your help!

In this case he had a old version of esptool.py, 2.1 version and update to a newer version, at that moment v2.5.

I'm using the last version available, the v2.6.
If I try to downgrade to 2.5 (the esptool.py) another error occurs, during the `make` command:
`esptool: error: unrecognized arguments: --elf-sha256-offset /home/user/esp/projects/hello_world/build/bootloader/bootloader.elf`

projectgus, 2019-03-20:
Hi @JoaoBerlese ,

Thanks for the detailed bug report.

One question: How big is bootloader.bin? Is it possible it's colliding with the partition table at 0x8000 if you also flashed this? (It doesn't mention in the bug report but I thought it was possible.) If so, you can [move the partition table](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-partition-table-offset) and rebuild everything.

If this doesn't help, can you please try updating to the latest master branch (v4.0-dev-91-gb7b5c3a8c or newer). [A bug was recently fixed in bootloader image generation](https://github.com/espressif/esp-idf/issues/3142) that may somehow be related to this.

> Similar problem on ESP32 forum Secure boot problem.

This is a little different because they're getting the "secure boot check fail" message: meaning, the secure boot digest for the bootloader image is incorrect. The error you have indicates the whole bootloader.bin image is somehow invalid (wrong checksum).

JoaoBerlese, 2019-03-21:
Hi @projectgus, thanks for the answer!

Works!

> One question: How big is bootloader.bin? Is it possible it's colliding with the partition table at 0x8000 if you also flashed this? (It doesn't mention in the bug report but I thought it was [...]

I changed the offset partition table (from 0x8000 to 0x10000) and everything finally worked!

#### Steps:
make menuconfig (command)
   Partition Table --->
[image]

Simple like that, thanks again. =]

projectgus, 2019-03-21:
Hi @JoaoBerlese ,

Thanks for updating, I'm glad this worked.

esptool.py will fail if a single flash command tries to flash overlapping files, but because the bootloader.bin is flashed in a separate step here it can't detect this. We have talked about adding a build system check for this problem (bootloader too big for partition table) we don't have this yet (clearly!)

closed 2019-03-22 by JoaoBerlese

driedler, 2020-02-20:
> One question: How big is bootloader.bin? Is it possible it's colliding with the partition table at 0x8000 if you also flashed this? (It doesn't mention in the bug report but I thought it was [...]

This was the issue for me. Logging the bootloader log level to 'error' resolved the issue.