espressif/esp-idf #3213: Memory corruption when using HTTP Client, if HTTP headers are big

MartinTJDK, 2019-03-21:
## Environment
IDF version: 3.1.3 (and older)

## Problem Description
SDK does not handle buffer overflow when creating http request.

File: esp_http_client.c
Method: esp_err_t esp_http_client_open(esp_http_client_handle_t client, int write_len)

[image]

Write of HTTP headers does not monitor for buffer overflow.
When writing a large Authentication header with e.g. 1KB token.
Of course it is possible to increase the HTTP buffer size, but it would be nice with a correct handling and reporting of ESP ERROR code.

### Expected Behavior
SDK will report error like:
```
   ESP_LOGE(TAG, "Out of buffer");
   return ESP_ERR_HTTP_CONNECT;
```

### Actual Behavior
No error is reported, and memory is corrupted.

### Steps to repropduce

1. Create a request with a 1KB header value
2. System will probably reboot (depending on what memory is overwritten)

mahavirj, 2019-05-16:
@MartinTJDK Sorry for slow turnaround. Is it possible for you to update to latest IDFv3.2 release (that has fix for this issue)? Fix is not in the form of clean patch, its part of rather big refactoring change, so it might be bit difficult to backport for v3.1 release.

mahavirj, 2019-07-21:
Closing due to lack of response.

closed 2019-07-21 by mahavirj