espressif/esp-idf #3214: esp_get_free_heap_size return wrong values

HumamHelfawi, 2019-03-21:
- Development Kit:      Geekworkm KE32-WROVER-C10
- Kit version 
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v3.1.1-6-g2aa9a2118
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:    Windows
- Power Supply:         USB

## Problem Description

`esp_get_free_heap_size `returns very huge amount  `1881206798 `. 

Related SDK config:
CONFIG_SPIRAM_SUPPORT=y

#
# SPI RAM config
#
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=
CONFIG_SPIRAM_USE_MEMMAP=
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_USE_MALLOC=
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_SIZE=4194304
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=
CONFIG_MEMMAP_TRACEMEM=
CONFIG_MEMMAP_TRACEMEM_TWOBANKS=
CONFIG_ESP32_TRAX=
CONFIG_TRACEMEM_RESERVE_DRAM=0x0

HumamHelfawi, 2019-03-22:
It was problem in wiring. I was controlling GPIO 16,17 which causes a lot of errors including this one.

closed 2019-03-22 by HumamHelfawi

KaeLL, 2022-05-11:
@HankLloydRight 
> The ESP32 pins GPIO16 and GPIO17 are not broken out to the ESP32-WROVER-B module and therefore not available for use. If you need to use these pins, please solder a module without PSRAM memory inside, e.g. the ESP32-WROOM-32D or ESP32-SOLO-1.

HankLloydRight, 2022-05-11:
My mistake, sorry.  The boards I'm using are  ESP32-WROVER, not  ESP32-WROVER-B. 
The  ESP32-WROVER  does have pin 16 broken out.