espressif/esp-idf #3215: MQTT disconnect message not implemented

gregjesl, 2019-03-21:
Duplicate issue of https://github.com/espressif/esp-mqtt/issues/97 (which does not appear to be tracked in your issue tracking database).  

## Issue
The MQTT client never sends the [DISCONNECT packet](http://public.dhe.ibm.com/software/dw/webservices/ws-mqtt/mqtt-v3r1.html#disconnect) to the broker. This results in the last will and testament always being sent, which is not the intended behavior.  

## Solution
I submitted the fix in https://github.com/espressif/esp-mqtt/pull/106

gregjesl, 2019-03-21:
@david-cermak I targeted the `idf` branch in my pull request but I branched from the `master` branch.  Should I branch from the `idf` branch for my pull request?

david-cermak, 2019-03-21:
@gregjesl Thank you for your help with this issue!

Yes please keep the `idf` as target branch (will soon merge it to master), but you seem to have to rebase onto idf.

gregjesl, 2019-03-29:
Closed per https://github.com/espressif/esp-mqtt/commit/caf5007b99df985b330683dfe2fa454c94633018

gregjesl, 2019-03-29:
Reopened - submodule needs to be updated.

mastertheknife, 2019-07-05:
Hi,
This is also the case with ESP8266, on the latest SDK (RTOS v3.2 - released 22 days ago)
As a temporary workaround for my project, i copied these changes into esp_mqtt_client_stop() and they work well.
I can confirm that this patch (and the subsequent one to only disconnect if connected: https://github.com/espressif/esp-mqtt/pull/118) work on the ESP8266:

```[kfir@joop esp-mqtt]$ diff -u mqtt_client.c.orig mqtt_client.c
--- mqtt_client.c.orig	2019-07-05 22:37:13.604686922 +0300
+++ mqtt_client.c	2019-07-05 22:53:15.787319054 +0300
@@ -788,6 +788,12 @@
 esp_err_t esp_mqtt_client_stop(esp_mqtt_client_handle_t client)
 {
     if (client->run) {
+        if(client->state == MQTT_STATE_CONNECTED) {
+            client->mqtt_state.outbound_message = mqtt_msg_disconnect(&client->mqtt_state.mqtt_connection);
+            if (mqtt_write_data(client) != ESP_OK) {
+                ESP_LOGE(TAG, "Error sending disconnect message");
+            }
+        }
         client->run = false;
         xEventGroupWaitBits(client->status_bits, STOPPED_BIT, false, true, portMAX_DELAY);
         client->state = MQTT_STATE_UNKNOWN;
```

closed 2019-07-25 by igrr