espressif/esp-idf #3217: [HID] How to answer incomming call with HID over BLE ?

Marc-Aurele, 2019-03-22:
Hello,

I would like to answer/decline a phone call with my ESP32. 
I know there is a Telephony usage page in the official HID usage table document but i'm not sure if i can use it to answer/decline a phone call. I also know the HFP profile should be more appropriate for this but i can't use it because there is no microphone on my board.
So, i'm likely to use the HID profil and the telephony usage page for my need but i'm not sure to come up with a solution. That's why i wanted to know if someone has already done this before (even if it needs to implement another profile over BLE and/or classic BT)

Best regards,

Aurélien

benjaminaigner, 2019-03-23:
There is a list for keyboard usage in Android:
https://source.android.com/devices/input/keyboard-devices.html#hid-consumer-page-0x0c

Maybe you need something like KEYCODE_HEADSETHOOK, but I'm not sure.

Depending on your implementation, you might need to add the "HID Consumer Page" to your HID report setup.

I didn't find anything worthy on iOS; but maybe the headset hook might work there as well.

Marc-Aurele, 2019-03-25:
Hi @benjaminaigner,

Thank you very much for the reply. If i'm understanding well, android does not interprete the telephony usage (cf. https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf). On the link you attached to your previous post, i can see : 
`
0x0c 0x01b6 | AL Image Browser | 0x00e2 | KEY_MEDIA | 3.0 | 0x004f | KEYCODE_HEADSETHOOK
-- | -- | -- | -- | -- | -- | --
`
So, i guess i need to add the following event : 0x01b6 (AL Image Browser ) in my consumer device page, then this event will be interpreted as  a "headset hook" on an android phone ?

Regards,

Aurélien

Marc-Aurele, 2019-03-27:
Hello,

I tried what you said but it didn't work. I tried to add a new report map in my current one : 

   ```
 0x95, 0x01,        // Report Count (1)
    0x05, 0x0B,        // Usage Page (Telephony)
    0x09, 0x01,        // Usage (Phone)
    0xA1, 0x01,        // Collection (Application)
    0x85, 0x03,        //   Report ID (3)
    0x09, 0x07,        //   Usage (Programmable Button)
    0xA1, 0x02,        //   Collection (Logical)
    0x05, 0x09,        //     Usage Page (Button)
    0x19, 0x01,        //     Usage Minimum (0x01)
    0x29, 0x06,        //     Usage Maximum (0x06)
    0x75, 0x03,        //     Report Size (3)
    0x15, 0x01,        //     Logical Minimum (1)
    0x25, 0x06,        //     Logical Maximum (6)
    0x81, 0x60,        //     Input (Data,Array,Abs,No Wrap,Linear,No Preferred State,Null State)
    0xC0,              //   End Collection
    0x05, 0x0B,        //   Usage Page (Telephony)
    0x09, 0x06,        //   Usage (Telephony Key Pad)
    0xA1, 0x02,        //   Collection (Logical)
    0x19, 0xB0,        //     Usage Minimum (Phone Key 0)
    0x29, 0xBB,        //     Usage Maximum (Phone Key Pound)
    0x25, 0x0C,        //     Logical Maximum (12)
    0x75, 0x04,        //     Report Size (4)
    0x81, 0x60,        //     Input (Data,Array,Abs,No Wrap,Linear,No Preferred State,Null State)
    0xC0,              //   End Collection
    0x05, 0x0B,        //   Usage Page (Telephony)
    0x09, 0x20,        //   Usage (Hook Switch)
    0x09, 0x29,        //   Usage (Alternate Function)
    0x09, 0x2C,        //   Usage (Conference)
    0x09, 0x25,        //   Usage (Transfer)
    0x09, 0x26,        //   Usage (Drop)
    0x09, 0x23,        //   Usage (Hold)
    0x09, 0x2B,        //   Usage (Speaker Phone)
    0x25, 0x07,        //   Logical Maximum (7)
    0x75, 0x03,        //   Report Size (3)
    0x81, 0x60,        //   Input (Data,Array,Abs,No Wrap,Linear,No Preferred State,Null State)
    0x05, 0x0C,        //   Usage Page (Consumer)
    0x09, 0xE0,        //   Usage (Volume)
    0x15, 0xFF,        //   Logical Minimum (-1)
    0x25, 0x01,        //   Logical Maximum (1)
    0x75, 0x02,        //   Report Size (2)
    0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,              // End Collection
```
But unfortunately, i didn't succeed to make it work.

Marc-Aurele, 2019-03-27:
Finally, the event play/pause can be used to accept an incoming call but i still don't know how to decline and i only tested it on android ...

benjaminaigner, 2019-04-23:
@Marc-Aurele 

Thanks for sharing this info.
Maybe the stop key works for declining?

Did you try this on iOS?

Marc-Aurele, 2019-04-23:
Hello @benjaminaigner 

Thank you very much for the help.
The STOP key didnt work on Android and i didn't test it on IPhone, because now i'm using the ANCS.
So, i didn't succeed to make the Telephony usage work so i gave up on using this and the PLAY/PAUSE for android and the ANCS for the Iphone is enough for what i need.

Regards,

Petros144, 2019-05-13:
You can use the Positive Action in the ANCS service. 

https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Specification/Specification.html

You can find it in the Apple Docs.

With an incomming call and for example new notification == 1  && Press_Button1:
 do ->EventFlagPositiveAction

 you can  accept a new call now.

Weijian-Espressif, 2019-07-05:
Hi, we have support ANCS , please update to the latest IDF and try, thanks.

Alvin1Zhang, 2019-12-04:
@Marc-Aurele Thanks for reporting, feel free to reopen with more feedback and details. Thanks.

closed 2019-12-04 by Alvin1Zhang

mak3rm1k3, 2020-04-21:
@Marc-Aurele exciting you were able to get the code working to answer the call, can you share some example code?

Thanks to @Petros144 for that example, but I'm new to arduino and don't know the syntax to implement the get Notification Attributes and get App Attributes, or the do Positive Action.

I'm trying to get notifications from iOS and ideally reply to some (like answer a call, send a reply text, etc.)