espressif/esp-idf #3219: ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE was not declared in this scope

PhilColbert, 2019-03-24:
Fresh install of esp idf and arduino core as from 23rd march.

Enabled all bluetooth serial and bluedroid options in menuconfig.

Compile fails with 

C:/msys32/home/Phil/esp/ble_uart/components/arduino/libraries/BluetoothSerial/sr                      c/BluetoothSerial.cpp: In function 'void esp_spp_cb(esp_spp_cb_event_t, esp_spp_                      cb_param_t*)':
C:/msys32/home/Phil/esp/ble_uart/components/arduino/libraries/BluetoothSerial/sr                      c/BluetoothSerial.cpp:161:34: error: 'ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE'                       was not declared in this scope
         esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
                                  ^
make[1]: *** [/home/Phil/esp/esp-idf/make/component_wrapper.mk:290: libraries/Bl                      uetoothSerial/src//BluetoothSerial.o] Error 1

Ideas ?

blueMoodBHD, 2019-03-27:
Hi @PhilColbert,  The API `esp_bt_gap_set_scan_mode` has been changed, please refer to commit `5beb2802e0b6ba3b533ca08ac78c13aa26d1fc2d`

Inespt, 2019-04-08:
Did you find a solution for this problem?

Inespt, 2019-05-16:
> @PhilColbert <https://github.com/PhilColbert> Thanks for reporting the
> issue. Has your issue been fixed? Thanks.

Yes

closed 2019-05-16 by Alvin1Zhang