espressif/esp-idf #3220: I2C slave not receiving broadcast

rustyx, 2019-03-24:
According to the [I2C spec](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) section 3.1.12, address 00 is a "general call" address, a.k.a. broadcast.

A write message sent to address 00 is supposed to be received by all slaves.

However, the ESP32 I2C slave seems to ignore it (I guess it's not implemented?).

Is there any workaround possible?

stickbreaker, 2019-03-24:
@rustyx if you set the slave id to 0 and engage slave mode, it will answer.  But, it will not answer as id 0 if slave id has a different value, or, if slave id is zero, but mode is master.  

It is a hardware limitation, the general call address is an optional feature of I2C.

Chuck.

stickbreaker, 2019-03-24:
@rustyx Although, you could use both I2C peripherals, program one for your normal address, the other as `zero` tie SCL0 to SCL1 and SDA0 to SDA1.   It would then be able to respond to both address zero and your other value.

Chuck.

rustyx, 2019-03-24:
I thought about that... but the other I2C is already used as a master, and besides I think a slave should not ACK a broadcast(?)

Anyway, it would be of immense help if there was a "sniff" slave mode, as I actually need to watch traffic between 4 different devices. Is there a place to file a hardware improvement suggestions?

stickbreaker, 2019-03-24:
@rustyx There is no problem with multiple slaves `ACK`ing the same address, each slave thinks it is the only one answering.  The first bit that does not match causes that slave to generate an internal `arbitration` error. 

I have successfully connected multiple 24LCxx EEPROMS with the same address and written to them.  They are perfectly happy to accept the same data.  If I read from them, the first one with a `0` bit where the other have a `1` bit wins then `arbitration` battle. 

One of the features of using the general call address is for error monitoring. (The slave hardware has to have this ability). Do a Read of address 0, the lowest id'd slave with error will answer, get that address, respond to error, rinse, repeat. When no one answers, all errors are cleared.

Chuck.

closed 2020-06-03 by Alvin1Zhang