espressif/esp-idf #3221: I2S: PCM Short Framesync, Clock (WS) inverted

mringwal, 2019-03-25:
## Environment

- Development Kit: SparkFun ESP32 Thing
- IDF version: v4.0-dev-71-g5beb2802e

## Problem Description

When configured for PCM Short Framesync in master mode, the Clock polarity is active-low instead of active-high, each sample is sent twice, and the data output starts one bit to early.

### Expected Behavior

I expect to see PCM Short Framesync on logic analyzer, similar to ESP32 Tech Ref Manual, Figure 53. WS is high-active, data starts after WS goes low.

[image]

### Actual Behavior

- Clock (WS) is active-low
- sample is output twice, 
- output starts one bit to early

[image]

### Steps to repropduce

Configure I2S for PCM Short, send data

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s.h"

#define NUM_SAMPLES 60
#define I2S_NUM         (0)
#define I2S_BCK_IO      (GPIO_NUM_26)
#define I2S_WS_IO       (GPIO_NUM_25)
#define I2S_DO_IO       (GPIO_NUM_22)
#define I2S_DI_IO       (-1)

void app_main()
{
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,                                  // Only TX
        .sample_rate = 8000,
        .bits_per_sample = 16,
        .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
        .communication_format = I2S_COMM_FORMAT_PCM | I2S_COMM_FORMAT_PCM_SHORT,
        .dma_buf_count = 6,
        .dma_buf_len = 60,
        .use_apll = false,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1                                //Interrupt level 1
    };
    i2s_pin_config_t pin_config = {
        .bck_io_num = I2S_BCK_IO,
        .ws_io_num = I2S_WS_IO,
        .data_out_num = I2S_DO_IO,
        .data_in_num = I2S_DI_IO
    };
    i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL);
    i2s_set_pin(I2S_NUM, &pin_config);
    i2s_set_clk(I2S_NUM, 8000, 16, I2S_CHANNEL_MONO);

    while (1) {
        uint16_t sample_data[NUM_SAMPLES];
        int i;
        for (i=0;i<NUM_SAMPLES/2;i++){
            sample_data[i*2]   = 0x5001;
            sample_data[i*2+1] = 0x5002;
        }
        size_t i2s_bytes_write;
        i2s_write(I2S_NUM, sample_data, 60 * 2, &i2s_bytes_write, portMAX_DELAY);
    }
}

koobest, 2019-04-12:
Hi,@mringwal 
The data out put  is not 1-bit early, but the WS signal is inverted. the WS  output signal can be reversed in gpio_matrix. You can try to modify i2s.c as follows:

[image]

My output signal:

[image]

thanks !!

tom-borcin, 2021-05-07:
Hi @mringwal,

we think the issue is resolved now and therefore we are closing it to keep our backlog manageable. If you don't agree with this decision, please let me know.

Thank you,

Tomas

closed 2021-05-07 by tom-borcin