espressif/esp-idf #3224: Error after Install the Required Python Packages

StasiukDanylo, 2019-03-25:
After installing requires I get:
> error: command 'i686-w64-mingw32-gcc' failed: No such file or directory
How I can fix this? 
Thank you.

[image]

dobairoland, 2019-03-25:
Hi @StasiukDanylo. Could you please post also the beginning of the log. I'm particularly interested in seeing which Python dependencies are not met. Also please post the output of `which python` command. Thanks!

ericd1949, 2019-04-02:
Excuse me from butting in, but I have also found this situation by following the "get-started" tutorial at:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html
I'm Windows 10 Pro, patched right up to date as of today. and I just followed the web instructions,
I'm pretty sure :-)

My python info:
$ python --version
Python 2.7.14

which python
/mingw32/bin/python

My path seems ok
$ echo $IDF_PATH
C:/msys32/home/erics/esp/esp-idf

As directed I used the python command, but it also failed in the same way if I used python2.7
I had tried it a few times, so some of the requirements were already met

Full log here and thanks in advance, Cheers, Eric

$ python -m pip install --user -r $IDF_PATH/requirements.txt
Requirement already satisfied: setuptools in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 4))
Requirement already satisfied: pyserial>=3.0 in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 8))
Requirement already satisfied: future>=0.15.2 in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from -r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 9))
Collecting cryptography>=2.1.4 (from -r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/07/ca/bc827c5e55918ad223d59d299fff92f3563476c3b00d0a9157d9c0217449/cryptography-2.6.1.tar.gz
Collecting pyparsing>=2.0.3 (from -r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/de/0a/001be530836743d8be6c2d85069f46fecf84ac6c18c7f5fb8125ee11d854/pyparsing-2.3.1-py2.py3-none-any.whl
Requirement already satisfied: asn1crypto>=0.21.0 in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Requirement already satisfied: six>=1.4.1 in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Requirement already satisfied: enum34 in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Requirement already satisfied: ipaddress in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Requirement already satisfied: pycparser in c:/msys32/home/erics/.local/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r C:/msys32/home/erics/esp/esp-idf/requirements.txt (line 10))
Installing collected packages: cryptography, pyparsing
  Running setup.py install for cryptography ... error
    Complete output from command C:/msys32/mingw32/bin/python.exe -u -c "import setuptools, tokenize;__file__='c:/temp/pip-build-gdzx1k/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:/temp/pip-pv_w27-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_py
    creating build
    creating build/lib.mingw-2.7
    creating build/lib.mingw-2.7/cryptography
    copying src/cryptography/exceptions.py -> build/lib.mingw-2.7/cryptography
    copying src/cryptography/fernet.py -> build/lib.mingw-2.7/cryptography
    copying src/cryptography/utils.py -> build/lib.mingw-2.7/cryptography
    copying src/cryptography/__about__.py -> build/lib.mingw-2.7/cryptography
    copying src/cryptography/__init__.py -> build/lib.mingw-2.7/cryptography
    creating build/lib.mingw-2.7/cryptography/hazmat
    copying src/cryptography/hazmat/_oid.py -> build/lib.mingw-2.7/cryptography/hazmat
    copying src/cryptography/hazmat/__init__.py -> build/lib.mingw-2.7/cryptography/hazmat
    creating build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/base.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/certificate_transparency.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/extensions.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/general_name.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/name.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/ocsp.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/oid.py -> build/lib.mingw-2.7/cryptography/x509
    copying src/cryptography/x509/__init__.py -> build/lib.mingw-2.7/cryptography/x509
    creating build/lib.mingw-2.7/cryptography/hazmat/backends
    copying src/cryptography/hazmat/backends/interfaces.py -> build/lib.mingw-2.7/cryptography/hazmat/backends
    copying src/cryptography/hazmat/backends/__init__.py -> build/lib.mingw-2.7/cryptography/hazmat/backends
    creating build/lib.mingw-2.7/cryptography/hazmat/bindings
    copying src/cryptography/hazmat/bindings/__init__.py -> build/lib.mingw-2.7/cryptography/hazmat/bindings
    creating build/lib.mingw-2.7/cryptography/hazmat/primitives
[... cut ...]
Command "C:/msys32/mingw32/bin/python.exe -u -c "import setuptools, tokenize;__file__='c:/temp/pip-build-gdzx1k/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:/temp/pip-pv_w27-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in c:/temp/pip-build-gdzx1k/cryptography/
You are using pip version 9.0.1, however version 19.0.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

dobairoland, 2019-04-02:
Hi @ericd1949,

Please try this command to resolve the issue:
`pacman -S mingw-w64-i686-python2-cryptography mingw-w64-i686-python2-pyparsing` 

If the previous one doesn't help then:
```
pacman -Sy pacman
pacman -Syu
```

ericd1949, 2019-04-03:
Thanks for your very swift response and suggestions, I'm swimming in the dark, a bit, but I'll give you the info I have.

I ran the first pacman command, and it re-installed pyparsing and cryptography, then I ran make flash
and it failed with:
**$ make flash**
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
The following Python requirements are not satisfied:
cryptography>=2.1.4
The recommended way ......

So I ran the last two pacman commands you gave me, then the first again and got the same failure when I tried to make flash.

I also ran:
**$ python -m pip install --user -r $IDF_PATH/requirements.txt**
and got the same failure as I gave in my first report.

Could it be that the cryptography version I have (1.9-1), is not sufficient for my apparent requirements
(2.1.4) ?

C:/msys32/home/erics/esp/esp-idf/requirements.txt           has cryptography>=2.1.4

and

$ **pacman -Ss cryptography**
mingw32/mingw-w64-i686-libbotan 2.3.0-1
    Botan is a C++ cryptography library released under the permissive Simplified BSD license (mingw-w64)
mingw32/mingw-w64-i686-python2-cryptography 1.9-1 [installed]
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw32/mingw-w64-i686-python3-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw64/mingw-w64-x86_64-libbotan 2.3.0-1
    Botan is a C++ cryptography library released under the permissive Simplified BSD license (mingw-w64)
mingw64/mingw-w64-x86_64-python2-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw64/mingw-w64-x86_64-python3-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)

I also tried the pacman suggestions given when make flash fails, so I've got yards (or meters) of logs,
let me know anything else you need.

Cheers,
Eric

dobairoland, 2019-04-03:
Yes @ericd1949, that is the problem. The installed package is too old. The following command I gave you before should update the packages:
```
pacman -Syu
```

Can you please post the output of this command?

ericd1949, 2019-04-03:
Ok, this is it:
**$ pacman -Syu**
:: Synchronising package databases...
 mingw32 is up to date                            0.0   B  0.00B/s 00:00 [---------------------------------------]   0%
 mingw64 is up to date                            0.0   B  0.00B/s 00:00 [---------------------------------------]   0%
 msys is up to date                               0.0   B  0.00B/s 00:00 [---------------------------------------]   0%
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

I tried doing a **make flash** again , just in case some change had happened, but it fails in the same way

Thanks.

dobairoland, 2019-04-03:
@ericd1949, I don't understand how this could be because after `pacman -Sy` the cryptography package should be of version 2.6. Your package database doesn't get updated for some reason. It seems to me some issue with MSYS2 and not IDF.

Can you try the MSYS2 ZIP file available from here:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#toolchain-setup
That should work.

ericd1949, 2019-04-03:
Ok I will do that. The file I had downloaded on 1st April was:
esp32_win32_msys2_environment_and_toolchain-20180110.zip
I'll let you know, there is some serious de-zipping to be done :-)

Cheers,
Eric

ericd1949, 2019-04-03:
Great, that fixed it - Hello World flashed and talking, changed a bit, re-compiled, and talking some more.
For the sake of info the MSYS file I downloaded was:
esp32_win32_msys2_environment_and_toolchain-20181001.zip
and having unzipped it to C: I just copied my home directory from my old msys32, as on the instructions.

Thanks for your quick responses and help, I may rest a bit before trying to get it to work in Eclipse :-)

Cheers,
Eric

dobairoland, 2019-04-03:
I'm happy that it got resolved.

How about you @StasiukDanylo? Did your issue got resolved as well?

mbermingham22, 2019-04-08:
Hello,
I'm having the same problem as described in this thread.  Except nothing listed here is clearing the "i686-w64-mingw32-gcc failed" error.  pacman doesn't think there is anything to do (i.e. everything is up-to-date).

mbermingham22, 2019-04-08:
If it helps, here's this:

$ pacman -S mingw-w64-i686-python2-cryptography mingw-w64-i686-python2-pyparsing
resolving dependencies...
looking for conflicting packages...

Packages (7) mingw-w64-i686-python2-asn1crypto-0.22.0-1  mingw-w64-i686-python2-cffi-1.10.0-1  mingw-w64-i686-python2-idna-2.6-1
             mingw-w64-i686-python2-pyasn1-0.2.3-1  mingw-w64-i686-python2-pycparser-2.17-1
             mingw-w64-i686-python2-cryptography-1.9-1  mingw-w64-i686-python2-pyparsing-2.2.0-2

Total Installed Size:  6.74 MiB

:: Proceed with installation? [Y/n] y
(7/7) checking keys in keyring                                                   [#############################################] 100%
(7/7) checking package integrity                                                 [#############################################] 100%
(7/7) loading package files                                                      [#############################################] 100%
(7/7) checking for file conflicts                                                [#############################################] 100%
error: failed to commit transaction (conflicting files)
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/__init__.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/__init__.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/_ast_gen.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/_ast_gen.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/_build_tables.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/_build_tables.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/_c_ast.cfg exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ast_transforms.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ast_transforms.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_ast.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_ast.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_generator.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_generator.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_lexer.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_lexer.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_parser.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/c_parser.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/lextab.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/lextab.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/__init__.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/__init__.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/cpp.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/cpp.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/ctokens.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/ctokens.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/lex.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/lex.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/yacc.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/yacc.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/ygen.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/ply/ygen.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/plyparser.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/plyparser.pyc exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/yacctab.py exists in filesystem
mingw-w64-i686-python2-pycparser: /mingw32/lib/python2.7/site-packages/pycparser/yacctab.pyc exists in filesystem
Errors occurred, no packages were upgraded.

dobairoland, 2019-04-08:
Hi @mbermingham22. The log shows that the version of your `cryptography` package is old (1.9). You need to update the packages by `pacman -Syu`.

The file "exists in filesystem" errors indicate that the given package cannot be installed by pacman because it was installed by pip. Try to remove it by `pip uninstall pycparser` and then install it by `pacman -S mingw-w64-i686-python2-pycparser`.

mbermingham22, 2019-04-08:
I tried that...here's what I had.  

$ pacman -Syu
:: Synchronizing package databases...
 mingw32 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 mingw64 is up to date                                    0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
 msys is up to date                                       0.0   B  0.00B/s 00:00 [---------------------------------------------]   0%
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

I'm trying to uninstall and reinstall pycparser, we shall see how that goes.

mbermingham22, 2019-04-08:
Okay, did all of that, and it keeps re-installing v1.9 of cryptography.  How do I force it to install the new one?

 pacman -Ss cryptography
mingw32/mingw-w64-i686-libbotan 2.3.0-1
    Botan is a C++ cryptography library released under the permissive Simplified BSD license (mingw-w64)
mingw32/mingw-w64-i686-python2-cryptography 1.9-1 [installed]
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw32/mingw-w64-i686-python3-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw64/mingw-w64-x86_64-libbotan 2.3.0-1
    Botan is a C++ cryptography library released under the permissive Simplified BSD license (mingw-w64)
mingw64/mingw-w64-x86_64-python2-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
mingw64/mingw-w64-x86_64-python3-cryptography 1.9-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)

 python -m pip install --user -r $IDF_PATH/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: setuptools in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 4)) (36.6.0.post20171017)
Requirement already satisfied: pyserial>=3.0 in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 8)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 9)) (0.17.1)
Requirement already satisfied: cryptography in c:/msys32/mingw32/lib/python2.7/site-packages (from -r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10)) (1.9)
Requirement already satisfied: idna>=2.1 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10)) (2.6)
Requirement already satisfied: asn1crypto>=0.21.0 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10)) (0.22.0)
Collecting six>=1.4.1 (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting enum34 (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/fc/d0/7fc3a811e011d4b388be48a0e381db8d990042df54aa4ef4599a31d39853/ipaddress-1.0.22-py2.py3-none-any.whl
Requirement already satisfied: cffi>=1.7 in c:/msys32/mingw32/lib/python2.7/site-packages (from cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10)) (1.10.0)
Requirement already satisfied: pycparser in c:/msys32/mingw32/lib/python2.7/site-packages (from cffi>=1.7->cryptography->-r C:/msys32/home/matt/esp/esp-idf/requirements.txt (line 10)) (2.17)
Installing collected packages: six, enum34, ipaddress
Successfully installed enum34-1.1.6 ipaddress-1.0.22 six-1.12.0

dobairoland, 2019-04-08:
Oh, I see. If these commands didn't help you then probably the only option is to download the new zipped MSYS2 environment: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#toolchain-setup.

naznaaa, 2019-04-11:
Hi,
I also get a similar error
log is
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/efuse/esp32/esp_efuse_table.o
CC build/bootloader/efuse/src/esp_efuse_fields.o
CC build/bootloader/efuse/src/esp_efuse_utility.o
CC build/bootloader/efuse/src/esp_efuse_api.o
AR build/bootloader/efuse/libefuse.a
LD build/bootloader/bootloader.elf
/bin/sh: pytho: command not found
make[1]: *** [/home/Nazna/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: /home/Nazna/esp/Smart_Warmer_as/build/bootloader/bootloader.bin] Error 127
make: *** [/home/Nazna/esp/esp-idf/components/bootloader/Makefile.projbuild:41: /home/Nazna/esp/Smart_Warmer_as/build/bootloader/bootloader.bin] Error 2

can you please hlep

dobairoland, 2019-04-11:
Hi @naznaaa. Please run `make menuconfig`, go into menu "SDK tool configuration" and make sure that the "Python 2 interpreter" is `python` and not `pytho`.

naznaaa, 2019-04-11:
Great,it's fixed

closed 2019-04-25 by dobairoland