espressif/esp-idf #3225: Unity causes make to fail with ESP-IDF Release v3.1.3 zip file install

freking, 2019-03-26:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.1.3 zip file
- Build System:         Make
- Compiler version:
   xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

the Unity directory is not found during the make process. 

### Expected Behavior
 make all or make flash to compile the code.   

### Actual Behavior
The command will return an error and does not finish compiling.
I no longer have the original error message.  
  
I fixed the issue by adding the submodule manually but I'm still receiving a warning.   
This is the warning.

WARNING: Missing submodule components/unity/unity...
Attempting 'git submodule update --init components/unity/unity' in esp-idf root directory...

I no longer have the original error message. 

After I added the submodule make works correctly

### Steps to repropduce

1. Unzip esp-idf-v3.1.3.zip file into the esp directory 
2. Copy an example application into esp 
3. make all 

### workaround 

Install Unity SubModule manually using the following commands 
$ git checkout master && git pull
$ cd ..
$ git add unity
$ git commit -m "updating unity submodule to latest"

projectgus, 2019-03-29:
Hi @freking ,

Thanks for the detailed bug report.

I think something quite unusual has happened, because there is no `components/unity` submodule in IDF v3.1.3. In v3.1.3 Unity is [not a submodule at all](https://github.com/espressif/esp-idf/tree/v3.1.3/tools/unit-test-app/components/unity). The unity submodule was added in commit 6091021e83d1046b9c393d90d2858d910de2b9b4 which is part of the IDF v3.3 prereleases and master branches, but not v3.1.x.

After thinking about this for a while, I think what may have happened is that the "example application" had a `build/` directory which was created from a newer version of ESP-IDF (which had this submodule). And the build directory included some build metadata files (`component_project_vars.mk`) which have paths for the submodules added in the newer IDF version. The first time the application is built with older IDF, it may incorrectly try to look for the newer submodule paths before the `component_project_vars.mk` files have been regenerated (and maybe there is even a fatal error before these paths can be regenerated).

I was unable to reproduce this in Linux but I'll try to reproduce it in Windows ASAP.

Do you think this may explain what you were seeing?

freking, 2019-03-31:
Yes, I think you are right. I did get some newer code for the
unit_test example buts it i no longer recall which version it was from
or the details of how I installed it . I thought this error was the
result of a problem with how the /unity/unity folder was labeled in
the Github repo.

projectgus, 2019-04-01:
> I did get some newer code for the unit_test example buts it i no longer recall which version it was from or the details of how I installed it

Ah, OK. If your project had some newer unit_test Makefile or component.mk in it then this may explain the problem entirely.

Do I understand correctly that you don't have a copy of the project which reproduces this behaviour any more? If so, maybe we can close this issue?

freking, 2019-04-01:
Yes,  I no longer have a copy of the project with this behavior . I
think its fine to close the issue.  I should have saved the project
when I encountered the issue. I don't remember the version of the
Unit_Test or if if came for a zip file or cloning the repo, so it may
be very hard to reproduce the issue. Since I had just rebuilt ESP-IDF
3.1.3 from the zip file I thought other users may get this error, but
I didn't think about the unit_test code I added.

projectgus, 2019-04-01:
OK, no problems. Thanks for explaining the circumstances.

closed 2019-04-01 by projectgus