espressif/esp-idf #3226: Modbus-RTU speed is related to temperature exceedingly

keelung-yang, 2019-03-26:
ESP32-PICO-KIT / ESP-IDF v3.3 / Windows 10 / Python 3.7.2 / PyModbus 2.1
-------------------------
The same hardware & software & configurations, sending 258 bytes(107 bytes per chunk) to esp32-pico-kit board by PyModbus:

At noon, take 124~125 us
At afternoon, take about 78~90 us
At 21:00, take 43~49 us

I threw my board to fridge at another noon to cool it, then tried again. At the first time, is slow as early, and after reboot my board, it run as quick as the best, but after running about 5 minutes, it slow down again.

alisitsyn, 2019-03-26:
Hi keelung-yang,

Could you try to measure response delay on your side using different tool (for example ModbusPoll)? What is the baudrate in your communication options? Did you try to play with Kconfig options? I can take a look to it later, but I need to reproduce your conditions on my side.

keelung-yang, 2019-03-27:
@alarruskain Thanks for your attention! 
There are unit typo: us should be ms(updated). 
The bautrate is 230400.
I've changed UART0 as the channel for Modbus-RTU, see attached sdkconfig.
I tried ModbusPoll, but seems no speed benchmark function, so I should write a simple script to simplify & reproduce this issue. Once I finished I will post it here.

At today noon, the time lap became about 60+ ms. So I brought a heat machine to heat my board until it's very hot by touching feeling. but the time lap has no obvious changes. It's weird.

[image]

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3011396/sdkconfig.zip)

alisitsyn, 2019-03-28:
Hi @keelung-yang,

The ModbusPoll has the Communication traffic feature (Display-Communication...) and there is Time stamp checkbox which will allow to get timestamps for request and response. I would ask you to check it there first to make sure the response delay is not related to PyModbus. Please note that the Modbus code has been updated to fix some known bug and also the new interface with master and slave support will be released in IDF v4.0. I expect the corrected error can be related to your issue. You can check the preliminary code here: https://www.esp32.com/viewtopic.php?f=13&t=9527&hilit=master+modbus

--
Alex

keelung-yang, 2019-04-06:
@alisitsyn It's just gone in recent days after going to master branch. I've to do that because large files(>7MB) on SD card can't opened in v3.3, but master works.

Since v3.3 changed too, and I've no local snapshot at that time. So I don't want to go back to figure out this. 

But I will take attentions to this, if it happens again, I'll reopen this issue.

closed 2019-04-06 by keelung-yang

alisitsyn, 2019-04-09:
@keelung-yang, 
Glad to hear that Modbus works. Yes, please reopen the issue once you get trouble again. I will try to help you from my side. Thank you for your attention to Modbus code and IDF. 

--
Alex