espressif/esp-idf #3227: what does it mean duration = 0?

yanzixiang, 2019-03-26:
in [test_rmt.c](https://github.com/espressif/esp-idf/blob/6f5820814fbd5e3f21dfbe6899259455f4f27621/components/driver/test/test_rmt.c) hast this testcase

```c
TEST_CASE("RMT tx set and get properties", "[rmt][test_env=UT_T1_RMT]")
{
    rmt_channel_t channel = RMT_TX_CHANNEL;
    uint8_t memNum;
    uint8_t div_cnt;
    bool loop_en;
    rmt_mem_owner_t owner;

    tx_init();
    TEST_ESP_OK(rmt_get_clk_div(channel, &div_cnt));
    TEST_ESP_OK(rmt_get_mem_block_num(channel, &memNum));
    TEST_ESP_OK(rmt_get_tx_loop_mode(channel, &loop_en));
    TEST_ESP_OK(rmt_get_memory_owner(channel, &owner));

    TEST_ASSERT_EQUAL_INT8(loop_en, 0);
    TEST_ASSERT_EQUAL_UINT8(div_cnt, RMT_CLK_DIV);
    TEST_ASSERT_EQUAL_UINT8(memNum, 1);
    TEST_ASSERT_EQUAL_INT(owner, RMT_MEM_OWNER_TX);

    //reset by "set"
    TEST_ESP_OK(rmt_set_pin(channel, RMT_MODE_TX, RMT_TX_GPIO_NUM));
    TEST_ESP_OK(rmt_set_clk_div(channel, RMT_CLK_DIV*2));
    TEST_ESP_OK(rmt_set_mem_block_num(channel, 2));
    TEST_ESP_OK(rmt_set_tx_loop_mode(channel, 1));
    TEST_ESP_OK(rmt_set_tx_carrier(channel, 0, 1, 0, 1));
    TEST_ESP_OK(rmt_set_idle_level(channel, 1, 0));
    TEST_ESP_OK(rmt_set_memory_owner(channel, RMT_MEM_OWNER_TX));

    TEST_ESP_OK(rmt_get_clk_div(channel, &div_cnt));
    TEST_ESP_OK(rmt_get_mem_block_num(channel, &memNum));
    TEST_ESP_OK(rmt_get_tx_loop_mode(channel, &loop_en));
    TEST_ESP_OK(rmt_get_memory_owner(channel, &owner));

    TEST_ASSERT_EQUAL_INT8(loop_en, 1);
    TEST_ASSERT_EQUAL_UINT8(div_cnt, RMT_CLK_DIV*2);
    TEST_ASSERT_EQUAL_UINT8(memNum, 2);
    TEST_ASSERT_EQUAL_INT(owner, RMT_MEM_OWNER_TX);

    rmt_item32_t items[1];
    items[0].duration0 = 300 / 10 * RMT_TICK_10_US;  //300us
    items[0].level0 = 1;
    items[0].duration1 = 0;
    items[0].level1 = 0;
    for(int i=0; i<100; i++) {
        TEST_ESP_OK(rmt_write_items(RMT_TX_CHANNEL, items,
                1, /* Number of items */
                1 /* wait till done */));
        vTaskDelay(10/portTICK_PERIOD_MS);  //every 10ms to write the item
    }
    TEST_ESP_OK(rmt_driver_uninstall(channel));
}
```

the TESTCASE configs the rmt_itme32_t like this,
what does it mean duration1 = 0?
```c
    rmt_item32_t items[1];
    items[0].duration0 = 300 / 10 * RMT_TICK_10_US;  //300us
    items[0].level0 = 1;
    items[0].duration1 = 0;
    items[0].level1 = 0;
```

[Is there any GUI to show the RMT data ?](https://github.com/espressif/arduino-esp32/issues/2593)

And what does it mean when both duration0 and duration01 are 0 ?
```
13:39:43.465 ->     "Index : 12 level0: 0 duration0: 1868 level1: 1 duration1: 993",
13:39:43.465 ->     "Index : 13 level0: 0 duration0: 1868 level1: 1 duration1: 2900",
13:39:43.465 ->     "Index : 14 level0: 0 duration0: 1868 level1: 1 duration1: 4332",
13:39:43.499 ->     "Index : 15 level0: 0 duration0: 1868 level1: 1 duration1: 3140",
13:39:43.499 ->     "Index : 16 level0: 0 duration0: 1868 level1: 1 duration1: 1945",
13:39:43.499 ->     "Index : 17 level0: 0 duration0: 1868 level1: 1 duration1: 4564",
13:39:43.499 ->     "Index : 18 level0: 0 duration0: 1876 level1: 1 duration1: 2652",
13:39:43.499 ->     "Index : 19 level0: 0 duration0: 1868 level1: 1 duration1: 4",
13:39:43.499 ->     "Index : 20 level0: 0 duration0: 4 level1: 1 duration1: 3132",
13:39:43.499 ->     "Index : 21 level0: 0 duration0: 1868 level1: 1 duration1: 0",
13:39:43.567 ->     "Index : 22 level0: 0 duration0: 44 level1: 0 duration1: 0",
13:39:43.567 ->     "Index : 23 level0: 0 duration0: 0 level1: 0 duration1: 0",
```

negativekelvin, 2019-03-26:
Duration 0 = END

koobest, 2019-03-26:
hi，
duration=0 means you want to stop the transmission. it always the last transmission data.

yanzixiang, 2019-03-27:
> hi，
> duration=0 means you want to stop the transmission. it always the last transmission data.

And what does this means ?

```C
13:39:43.567 ->     "Index : 23 level0: 0 duration0: 0 level1: 0 duration1: 0"
```

negativekelvin, 2019-03-27:
level0: 0 duration0: 0 this means END
level1: 0 duration1: 0 this we don't care because we already ENDED

closed 2019-03-29 by yanzixiang