espressif/esp-idf #3228: CMake and make usage causes full rebuild

jeremyherbert, 2019-03-28:
I'm currently swapping between cmake (for the IDE support) and make (for the flash encryption/secure boot). Unfortunately, every time the IDE refreshes the CMake configuration, it causes the sdkconfig to be modified (deleting a bunch of serial flasher keys). This means that the entire project needs to be rebuilt every time I refresh the CMakeLists.txt config.

Is there a plan to support both in the future? I'd like to move completely over to CMake, but it seems that this project I am working on will be stuck with a mix of both.

projectgus, 2019-03-28:
Hi @jeremyherbert ,

You may know this already, but the root cause of what you're experiencing is that the serial flasher settings and the Python interpreter setting were moved out of KConfig in the CMake-based build system (see #1539), so - as you've noticed - these keys only appear if sdkconfig is generated by "make" not by "CMake".

In general, we hadn't really anticipated people may try to use both build systems side by side at the same time with the same project. So this is not supported, and probably won't be supported.

> Is there a plan to support both in the future? 

We're switching to CMake-based build system as the default in IDF V4.0, and the plan is to finish the missing build system features (secure boot, flash encryption, Eclipse support) before then, so there is feature parity.

For IDF v4.x we will continue to support GNU Make based build system as well. Although probably in an "either/or" scenario rather than "use both at the same time".

For now, the best recommendation I can give you is to stick to GNU Make for a little while longer until you're able to update to a version which has CMake support for the features you need. We should reach feature parity in the master branch in the next month or so, although we don't have a date for a V4.0 release yet.

jeremyherbert, 2019-03-28:
Ok, no problems. I desperately look forward to secure boot support in CMake ;)

closed 2019-03-28 by jeremyherbert

projectgus, 2019-03-28:
@jeremyherbert I did just think of a workaround which might work, which is that you can override the name of sdkconfig by setting SDKCONFIG in either the Makefile or the CMakeLists.txt file (or by passing a variable to idf.py )

So if you used two side-by-side sdkconfig files, one for GNU Make and one for CMake, then probably everything would work. You'd just need to remember to sync any changes between them.

But we'll get the secure boot support in CMake ASAP, so this kind of unpleasantness is not required. :)

jeremyherbert, 2019-03-28:
Ok great. My workaround was to `chmod -w sdkconfig`, but your way is nicer I think

projectgus, 2019-03-28:
Oh, nice! That one has the advantage that you can't get the sdkconfig files out of sync.

jerry-weng, 2019-03-30:
@projectgus Here is my solution to use shell glob pattern for flash command.
```shell
diff --git a/components/esptool_py/Makefile.projbuild b/components/esptool_py/Makefile.projbuild
index 25f4b8e6f..7157513de 100644
--- a/components/esptool_py/Makefile.projbuild
+++ b/components/esptool_py/Makefile.projbuild
@@ -1,6 +1,6 @@
 # Component support for esptool.py. Doesn't do much by itself,
 # components have their own flash targets that can use these variables.
-ESPPORT ?= $(CONFIG_ESPTOOLPY_PORT)
+ESPPORT ?= $(shell echo $(CONFIG_ESPTOOLPY_PORT))
 ESPBAUD ?= $(CONFIG_ESPTOOLPY_BAUD)
 ESPFLASHMODE ?= $(CONFIG_ESPTOOLPY_FLASHMODE)
 ESPFLASHFREQ ?= $(CONFIG_ESPTOOLPY_FLASHFREQ)
```
then, we can set `CONFIG_ESPTOOLPY_PORT="/dev/tty.usbserial*"` in `sdkconfig` file.

projectgus, 2019-03-31:
@jerry-weng That's a clever solution, but it will produce an error if there is more than one usb/serial converter attached to the system.

Actually, since v2.6 esptool.py has the ability to scan for a serial port with an attached chip. We don't expose this behaviour in the GNU Make based build system, but it would be possible to by removing the `-p $(ESPPORT)` argument entirely.

If you use `idf.py` and the [CMake-based build system](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/), it will automatically pick the first serial port that it finds. This is our long term plan for the IDF build system.

jerry-weng, 2019-04-02:
Yes, if more than one usb/serial converter is attached, specifing `ESPPORT` manually is a better solution.