espressif/esp-idf #3229: Heap corruption with a large malloc

kosha-ve, 2019-03-28:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version : v4.0-dev-78-g8fbb63c2a
- Build System:         [Make]
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:     [macOS & Linux]
- Power Supply:         [external 3.3V]

## Problem Description

I see heap memory corruption when I try to allocate a somewhat larger buffer (>548 bytes). 
The function I use for malloc is - heap_caps_malloc(sz, MALLOC_CAP_8BIT);
I printed out each malloc (start and end address) and free along with free heap size, and heap intergrity check.

Example log1:
malloc 0x3ffddffc - 0x3ffde038
malloc 0x3ffdcadc - 0x3ffdce80
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffdce80
abort() was called at PC 0x400907cb on core 0

Example log2:
malloc 0x3ffdc4a8 - 0x3ffdc4e4, free mem - 127508
malloc 0x3ffdf734 - 0x3ffdfad8, free mem - 126572
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffdfad8
abort() was called at PC 0x400907cb on core 0

Note: There are no calls to free() between the two mallocs.

### Expected Behavior
malloc to allocate required buffer correctly or else throw error.

### Actual Behavior

In the first example, the second memory allocation overlaps the first one.
And in the second example, the memory available actually increases after doing a large malloc.
This causes heap corruption and hence abort().

igrr, 2019-03-28:
Hi @kosha-ve, could you please post the code which produces this issue?

> In the first example, the second memory allocation overlaps the first one.

    malloc 0x3ffddffc - 0x3ffde038
    malloc 0x3ffdcadc - 0x3ffdce80
    CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffdce80

The order of the 4 values above is: 0x3ffdcadc, 0x3ffdce80, 0x3ffddffc, 0x3ffde038. So the 1st and 2nd ranges do not overlap. Is your code doing something with these 2 buffers (writing to them), or is it only allocating the buffers?

> And in the second example, the memory available actually increases after doing a large malloc.

    malloc 0x3ffdc4a8 - 0x3ffdc4e4, free mem - 127508
    malloc 0x3ffdf734 - 0x3ffdfad8, free mem - 126572
    CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffdfad8
    abort() was called at PC 0x400907cb on core 0

Here the free memory printed in the 2nd line is smaller than in the first one, so I don't understand what you mean by "memory available actually increases after doing a large malloc". Also the failure here has the same pattern: the word after the end of the 2nd buffer is corrupted.

Is it possible that you are calculating `end = buffer + size`, and then writing data to `end`, inclusive? 

Or, that `buffer` is a pointer to something that is not `uint8_t` or `char`, in which case `end = buffer + size` ends up being `end = ((uint32_t) buffer) + size * sizeof(buffer[0])`?

kosha-ve, 2019-03-28:
I try to read a file from spiffs into that buffer.
``static size_t _read_file(const char filename[], char **buf)
{
    char *pPtr = NULL;
        size_t len = -1;
    FILE *fp = NULL;

    *buf = NULL;

        /* check if file exists */
    struct stat st;
    if ((NULL != buf) && (stat(filename, &st) == 0))
    {
                fp = fopen(filename, "r");
                if (NULL != fp)
                {
                    printf("reading file - %s\n", filename);
                        pPtr = (char *)heap_caps_malloc(st.st_size, MALLOC_CAP_8BIT);
                        if (NULL == pPtr)
            {
                fclose(fp);
                goto exit;
            }

            len = fread(pPtr, 1, st.st_size, fp);
            if (len == st.st_size)
            {
                *buf = pPtr;
            }
            else
            {
                heap_caps_free(pPtr);
                len = -1;
            }
            fclose(fp);
        }
        }

exit:
        return len;
}``

These buffers are not further written into, but only read from.

> The order of the 4 values above is: 0x3ffdcadc, 0x3ffdce80, 0x3ffddffc, 0x3ffde038. So the 1st and 2nd ranges do not overlap. Is your code doing something with these 2 buffers (writing to them), or is it only allocating the buffers?

I agree that the observations are wrong. But the corruption happens only when I allocate a larger buffer. If the buffer is smaller, it works fine.

kosha-ve, 2019-03-28:
@igrr Thanks for looking into my problem. I have been running more debug at my end and I think I may have found the issue in my code. Will close the issue once I run it a couple more times and am sure of the issue.

closed 2019-03-28 by kosha-ve

Sarah-C, 2021-07-23:
How can a malloc go wrong in your code?

Did you find the solution? I've got the same problem.