espressif/esp-idf #3230: [HID] HID over BLE - Has someone already implemented ANCS on ESP32 ?

Marc-Aurele, 2019-03-28:
Hello,

Has someone already implemented ANCS on ESP32 ?
I would like to implement this feature because i need to answer/decline phone call on iphone, do you think it is possible ?

Regards,

Aurélien

chegewara, 2019-03-28:
Its arduino library, but as far as i know it did work:
https://github.com/S-March/esp32_ANCS

Marc-Aurele, 2019-03-28:
Hi @chegewara 
Thank you for the link. But even with this i'm a little bit lost and i don't know how to port this arduino library on my project. 
Do you think i can start with the BLE Gatt client example to be able to connect with a BLE ANCS server ?

Marc-Aurele, 2019-04-01:
I'm trying to implement the example you gave to me. I'm launching a gatt server, once the iphone is connected to this server (when i receive ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT) i launch a GATT client in order to be connect with the ANCS server but on the client side, i'm able to connect with the server (i receive the ESP_GATTC_CONNECT_EVT event) but i cannot open the connection : i don't receive the ESP_GATTC_OPEN_EVT event.
To open the connection, i launch :  
```
errRc = esp_ble_gattc_open(m_gattc_if,  &m_address, BLE_ADDR_TYPE_PUBLIC, 1 );
```
But after a certain period of time i receive : 
```
W (38681) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0100W (38681) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0100
I (38681) BLE_TASK: ESP_HIDD_EVENT_BLE_DISCONNECT
I (38681) BLE_ANCS ---> : ESP_GATTC_DISCONNECT_EVT
I (38691) BLE_ANCS ---> : ESP_GATTC_DISCONNECT_EVT, reason = 256
I (38691) BLE_ANCS ---> : ESP_GATTC_OPEN_EVT
E (38701) BLE_ANCS ---> : open failed, status 133
```

chegewara, 2019-04-02:
From arduino code i see its just connecting as client to iPhone when iPhone is connected to esp32.
My only suggestion is to check if you have setup all security options, it may be the case, because apple is crazy about it.

Marc-Aurele, 2019-04-02:
I think i'm using the same hehavior as the one for arduino. Indeed, im launching a server and i connect my iphone with it, once is connect, i launch a client and i try to connect to the ANCS server; we are agree i don't need to scan ? Is it supposed to work with iphone5 ?

chegewara, 2019-04-02:
I dont know about iPhone, but you could try and flash this arduino project to your esp32. Then you can see if it works with iPhone 5 and maybe see logs too.

Marc-Aurele, 2019-04-02:
I tried to used Arduino to compile and load this library but i did'nt succeed to make it compiled
- I have installed Arduino IDE 1.8.9
- I have added some boards with the board manager (im using DOIT dev kit V1)
- I have installed the esp-idf for arduino (i can see it in ~/Arduino/hardware/espressif)
- I'm abled to compile ESP32 examples and to load them
It is the first time i use Arduino so i don't really know how to load the github project of ANCS properly ...

chegewara, 2019-04-02:
Sorry, you are right. esp-idf and arduino ble library changed a lot since this ANCS code has been made.
Try with this code. I changed it and it is compiling now, also you need to get src folder from this zip and replace folder in `~/libraries/BLE/`

[esp32_ANCS-master.zip](https://github.com/espressif/esp-idf/files/3036249/esp32_ANCS-master.zip)

If you have more questions i will try to help.

Marc-Aurele, 2019-04-03:
Hello,

Thank you very much @chegewara 
I'm able now to compile and load on my ESP32. I have tested on my iphone 5 and it is working even if sometimes i can see some stack issue that make my board to be reset. I cannot pair anymore my ESP32 with my android phone. But  with the iphone, I can see incoming call ...
Before going further, can you confirm it is possible to pick up / decline phone call with ANCS ?

Best regards,

Aurélien

chegewara, 2019-04-03:
This question is to @S-March i think.

Marc-Aurele, 2019-04-03:
It's OK i finally found how to pickup a phone call. Thx again @chegewara and @S-March for this work.
Now, i ll try to port this arduino library in my C project.

Marc-Aurele, 2019-04-04:
I have one more question. If i want to launch the ANCS server on IOS side, i need to advertise with specific data that embeds the ANCS server UUID.  If i connect with an Iphone, thanks to the advertising data, the iphone can see it needs to launch an ANCS server. Am i right ? But are these advertising packets are compatible with android ?

Marc-Aurele, 2019-04-04:
Here is the advertising data i'm using : 

```
static uint8_t hidd_adv_data_raw[] = {
        /* flags */
        0x02, ESP_BLE_AD_TYPE_FLAG, 0x01,
        /* service uuid */
        17, ESP_BLE_AD_TYPE_128SOL_SRV_UUID, 0xD0, 0x00, 0x2D, 0x12, 0x1E, 0x4B, 0x0F, 0xA4, 0x99, 0x4E, 0xCE, 0xB5, 0x31, 0xF4, 0x05, 0x79,
};
```
But when i'm using this, i cannot pair anymore with my device from an android phone ...

Weijian-Espressif, 2019-04-10:
@Marc-Aurele , we will add ble_ancs demo later, please wait a while, thanks.

Marc-Aurele, 2019-04-10:
Hello @Weijian-Espressif 

I finally make the ANCS client work in my project, it is not so complicated but the project from @S-March helped me a lot.
Now, i'm able to combined my HID profil with ANCS and to send the right events according the phone paired to my device.

nagup, 2020-01-01:
Multiple libraries were found for "Task.h"
 Used: C:\Program
Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\..\Documents\Arduino\libraries\ESP32_BLE_Arduino-master
 Not used: C:\Program
 Not used: C:\Users\..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE
exit status 1
'class BLEAdvertisementData' has no member named 'setServiceSolicitation'

can someone help me with this error please. I'm new to ArduinoIDE

chegewara, 2020-01-02:
> can someone help me with this error please. I'm new to ArduinoIDE

You are new to arduino, you should open issue here:
https://github.com/espressif/arduino-esp32

nagup, 2020-01-02:
I have already got everything I need to use the ESP32 with Arduino IDE, I have uploaded many examples and sketches online to my ESP32, I'm just having a problem with this sketch

nagup, 2020-01-02:
Now I have this message

[New Text Document (2).txt](https://github.com/espressif/esp-idf/files/4015894/New.Text.Document.2.txt)

nagup, 2020-01-18:
I have figured it out, it was so simple but no one told me that I need to put all those .h and .cpp files in the same folder as the arduino sketch. Well I've told whoever will need this in the future. If you need more help comment under this, send me a message or something. I am more than happy to help you more than just saying a few words and letting you find out the rest.

Thank you!

chegewara, 2020-01-19:
@nagup if you would open issue in arduino repository you would get more help probably.

nagup, 2020-01-19:
@chegewara I've figured it out, but I need help on finding out what part of the code deals with the notification. 

I am after the actual text in a notification like "Instagram: User_123 Liked your post" not only "Category: Social, Entertainment, Other" because I want to display the text string on a display.

Can you tell me what part of the code I can find this? First I want to print this string in serial monitor so just tell me roughly where to look and I'll try figure out the rest.

Thank you