espressif/esp-idf #3231: Will ESP-IDF upgrade to the Amazon FreeRTOS?

maxgerhardt, 2019-03-28:
Currently ESP-IDF uses 

https://github.com/espressif/esp-idf/blob/76da10a665c228b81225130ca2060d04c1ee000a/components/freertos/include/freertos/FreeRTOS.h#L1-L3

Now the people at https://github.com/aws/amazon-freertos have released documentation and demos regarding Amazon FreeRTOS on the ESP32 (https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html and demo code https://github.com/aws/amazon-freertos/tree/master/demos/espressif/esp32_devkitc_esp_wrover_kit). 

This uses 

https://github.com/aws/amazon-freertos/blob/5b2967a59ecf8b3a1385bca0797ad93842d9bee3/lib/include/FreeRTOS.h#L1-L4 

The demo project seems kind of weird to me, because the [components folder](https://github.com/aws/amazon-freertos/tree/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code) agains adds an `freertos` folder (that doesn't interfere with Espressif's built-in `freertos` component) and lots of other already existing components like `tcpip_adapter` etc., which apparently had to be adapter to work with Amazon FreeRTOS. 

Will ESP-IDF switch to the Amazon FreeRTOS in the future or will it continue to use FreeRTOS V8 from 2015?

negativekelvin, 2019-03-28:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html

maxgerhardt, 2019-03-28:
@negativekelvin The documentation states that FreeRTOS v8.2.0 was used as a base with a few backported features from V9.0.0, and while this documentation of the implementation and designs is interesting it doesn't answer the question stated in the issue; thanks regardless.

negativekelvin, 2019-03-28:
The most important part is 
>The ESP-IDF FreeRTOS is a modified version of vanilla FreeRTOS which supports symmetric multiprocessing (SMP)

>Currently, the Amazon FreeRTOS port for ESP32-WROVER-KIT and ESP DevKitC does not support the following features:
Lightweight IP.
Symmetric multiprocessing (SMP).

So unless you are using esp32-solo you lose dual core.  If there is a specific feature you are looking for from V10 then it would be useful to mention it.  You could also encourage Amazon to add support for SMP.

mahavirj, 2019-03-29:
@maxgerhardt 

> Now the people at https://github.com/aws/amazon-freertos have released documentation and demos regarding Amazon FreeRTOS on the ESP32 

Port for ESP32 in Amazon FreeRTOS is maintained by Espressif.

> The demo project seems kind of weird to me, because the components folder agains adds an freertos folder (that doesn't interfere with Espressif's built-in freertos component) and lots of other already existing components like tcpip_adapter etc., which apparently had to be adapted to work with Amazon FreeRTOS.

Extra `freertos` component directory is required because of compatibility with IDF,
 - IDF expects freertos header to have include path of sort `freertos/task.h`
 - Some extra [stubs](https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/freertos/extra.c) needs to be provided to work with vanilla freertos

Other duplicated components are for similar reasons. Regarding `tcpip_adapter` component, it is adapted to support FreeRTOS TCP stack (IDF uses lwIP networking stack).

> Will ESP-IDF switch to the Amazon FreeRTOS in the future or will it continue to use FreeRTOS V8 from 2015?

This largely depends on SMP support in vanilla freertos. Interim, any specific requirement that your are looking for?

ammaree, 2019-03-29:
Here is an example of the functionality missed... #1453

maxgerhardt, 2019-03-29:
Thanks @ammaree for the issue link, I think @projectgus has answered my question there already:

> We've decided to stop backporting FreeRTOS features to our ESP-IDF fork of FreeRTOS V9, and instead we're focusing on updating to add SMP support to FreeRTOS V10.
>
>So far we don't have an ETA for this update, I'm sorry. We will update this GitHub issue as soon as that work is done, though.

closed 2019-03-29 by maxgerhardt

crippsJack, 2019-08-14:
Hi, is there any ETA for adding SMP support to FreeRTOS V10 yet?

Thanks,

p0wl, 2020-09-11:
any updates on this?

projectgus, 2020-09-14:
Hi @p0wl,

There is a preview branch of ESP-IDF running FreeRTOS 10 here:
https://github.com/espressif/esp-idf/tree/feature/freertos_upstream_plus_tlsf_allocator_preview

You can read more about it here:
https://esp32.com/viewtopic.php?f=10&t=15322

There is still no upstream API for SMP in Amazon FreeRTOS, so this branch still relies on patches to add SMP to some APIs compared to the official FreeRTOS APIs.