espressif/esp-idf #3232: Will ESP32 support EAP-TLS authentication method in the future?

elitezhu, 2019-03-29:
Now our team are trying ESP32 module into our concepted sensors which deliver the data to Mindsphere directly. You konw many factories have the secure management of wireless. According to the ESP-IDF SDK, I can succeed to connect the kind of sensor to our secure AP with the authentication method of EAP-PEAP which needs username and password. In the real factory plant, it's barely convenient.  But the EAP-TLS does not need this kind of info, it just need the certificate which is more possible to deploy the sensors with ESP32 module in factory plant via secure wireless.

So will Espressif have a plan to support EAP-TLS in the future?
Deeply hope it comes out.

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

liuzfesp, 2019-03-29:
Hi @elitezhu IDF already supports EAP-TLS/EAP-PEAP/EAP-TTLS.

elitezhu, 2019-03-29:
@liuzfesp , 
Thanks for replying.
But in the example/wifi/wpa2_enterprise, the SDK just define the EAP-PEAP and EAP-TTLS micro which I can refer and configure in menuconfig. So if I choose EAP-TLS, what's the micro number I need configure   in the menuconfig.

negativekelvin, 2019-03-29:
https://github.com/espressif/esp-idf/blob/master/examples/wifi/wpa2_enterprise/main/Kconfig.projbuild#L14

Alvin1Zhang, 2019-04-04:
@negativekelvin Thanks for the updates, @elitezhu has your issue been resolved? Thanks.

elitezhu, 2019-04-04:
@Alvin1Zhang @negativekelvin Sorry，I have returned back from BeiJing, so I can't verify it right now. But I think I have got the right answer. Thanks for replying.
And the issue can be closed.

Alvin1Zhang, 2019-04-04:
Thanks @elitezhu Feel free to reopen if any updates, thanks.

closed 2019-04-04 by Alvin1Zhang