espressif/esp-idf #3233: I got message : Unresolved inclusion: "esp_event.h", I can't find "esp_event.h" file in esp-idf/components/esp32/include, please help

zhouypau, 2019-03-30:
I got message : Unresolved inclusion: "esp_event.h", I can't find "esp_event.h" file in esp-idf/components/esp32/include, please help

negativekelvin, 2019-03-30:
https://github.com/espressif/esp-idf/commit/fd9bc30aca5d76a57f72ae2d0fa6181205b5811d

negativekelvin, 2019-03-30:
https://github.com/espressif/esp-idf/commit/5a83347bec398b202570332061d6100f26abb51d
> components/esp32/include/esp_event.h → components/esp32/include/esp_event_legacy.h

> components/esp_event/include/esp_event.h

zhouypau, 2019-03-30:
Thanks, Great ! the esp_event.h resolved !
BUT after I clean and rebuild, I got erro:
Error: build command 'cmake' not FoundFailure running cmake

therefore I still can't get ' esp_err_t ' 'system_event_t' 'ESP_ERROR_CHECH()...... to be resolved

zhouypau, 2019-03-30:
eclipse terminal messages:

Building in: /home/yiping/EclipseProject/mybpp/build/default
cmake -G Ninja -DCMAKE_BUILD TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMAMDS=ON /home/yiping/EclipseProject/mybpp
Error: build command 'cmake' not foundFailure running cmake:

zhouypau, 2019-03-31:
add components/esp_event/include to PATH , solved issue

closed 2019-03-31 by zhouypau