espressif/esp-idf #3234: Wifi: SYSTEM_EVENT_AP_STADISCONNECTED  sometimes doesn't fire

lucascoxBAF, 2019-03-30:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-197-gda9096682
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-ng-1.22.0-73-ge28a011
- Operating System:     [Linux]

## Problem Description

I've noticed some odd behavior when running in AP mode. Every once in a while, no SYSTEM_EVENT_AP_STADISCONNECTED will be fired when a client disconnects from an AP. This causes issues because I start tcp server and an mdns sever when I get a connect event and the connection count is currently 0, and I stop both of them when I get a disconnected event and the count is (now) 0. 

### Expected Behavior
Every time a client disconnects, a SYSTEM_EVENT_AP_STADISCONNECTED will fire.

### Actual Behavior
Most of the time, the event fires, but some times it does not. 

## Debug Logs

In the following log, I connected to the ap, disconnected, connected again, and disconnected again. Not that there are two connected events, but only one disconnected event. 

```
I (8992) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (8992) wifi: station: 68:db:ca:34:46:4a join, AID=1, bgn, 20
I (9007) ESP Wifi: **SYSTEM_EVENT_AP_STACONNECTED**
I (9011) Http Server: Starting server on port: '31416'
I (9215) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (9237) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (9237) Ble Driver: Stop adv successfully

I (9238) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (9238) Ble Driver: scan start success
W (9513) resend!: 
W (10013) resend!: 
I (10238) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (10238) Ble Driver: scan stop success
I (10238) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (10240) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (10240) Ble Driver: Advertising Succeeded!
W (10513) resend!: 
W (11013) resend!: 
I (11239) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (11239) Ble Driver: Stop adv successfully

I (11240) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (11240) Ble Driver: scan start success
W (12013) resend!: 
I (12239) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (12240) Ble Driver: scan stop success
I (12240) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (12242) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (12242) Ble Driver: Advertising Succeeded!
W (12513) resend!: 
W (13013) resend!: 
I (13241) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (13241) Ble Driver: Stop adv successfully

I (13242) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (13242) Ble Driver: scan start success
W (13513) resend!: 
I (14241) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (14242) Ble Driver: scan stop success
I (14242) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (14244) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (14244) Ble Driver: Advertising Succeeded!
I (14844) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (14844) wifi: station: 68:db:ca:34:46:4a join, AID=1, bgn, 20
I (14919) ESP Wifi: **SYSTEM_EVENT_AP_STACONNECTED**
I (15041) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (15243) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (15243) Ble Driver: Stop adv successfully

I (15244) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (15244) Ble Driver: scan start success
I (16244) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (16244) Ble Driver: scan stop success
I (16244) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (16247) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (16247) Ble Driver: Advertising Succeeded!
W (16731) resend!: 
W (17231) resend!: 
I (17246) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (17246) Ble Driver: Stop adv successfully

I (17247) Ble Driver: event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
I (17247) Ble Driver: scan start success
W (17731) resend!: 
W (18231) resend!: 
I (18247) Ble Driver: event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
I (18247) Ble Driver: scan stop success
I (18247) Esp Ble Server: advertisingParams: 32, 64, 7, 0.
I (18249) Ble Driver: event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
I (18249) Ble Driver: Advertising Succeeded!
I (18870) wifi: station: 68:db:ca:34:46:4a leave, AID = 1, bss_flags is 134243
I (18870) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (18871) ESP Wifi: **SYSTEM_EVENT_AP_STADISCONNECTED**
I (19247) Ble Driver: event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
I (19248) Ble Driver: Stop adv successfully
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

liuzfesp, 2019-03-31:
HI @lucascoxBAF how about wait for more than 5 minutes before connect the ESP32 again?

lucascoxBAF, 2019-04-01:
@liuzfesp, After waiting five minutes, the inactivity timer expires and the disconnect event fires. Is it expected that disconnect events will periodically be missed by the esp32 and I'll need to wait five minutes before the inactivity event fires? Or is this a bug that is being worked on?

liuzfesp, 2019-04-04:
Hi @lucascoxBAF the missed SYSTEM_EVENT_AP_STADISCONNECTED can happen in following scenario:

- The station connects to ESP32 soft-AP
- The station disconnects WiFi, but it failed to send deauth/disassociation to ESP32 soft-AP, or the deauth/disassociation frame is not successfully received by ESP32 soft-AP
- The ESP32 soft-AP doesn't observed that the station is disconnected until the keepalive timeout (5 minutes)

lucascoxBAF, 2019-04-09:
@liuzfesp, thanks for the info. I was surprised at how reproducible this is, though. It seems be directly related to cpu load. If the cpu load is higher, the disconnect events are missed much more frequently. However, if this isn't a bug, I can close it.

zhangyanjiaoesp, 2020-01-19:
@lucascoxBAF 
I think the explanation of @liuzfesp is correct, and this is not a problem. You can close this issue.

Alvin1Zhang, 2020-05-29:
Thanks for reporting, feel free to reopen or file a separate ticket for new issue. Thanks.

closed 2020-05-29 by Alvin1Zhang