espressif/esp-idf #3235: MDNS: services are only being advertise to first connection in AP mode

lucascoxBAF, 2019-03-30:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-197-gda9096682
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-ng-1.22.0-73-ge28a011
- Operating System:     [Linux]

## Problem Description

It appears that the mdns service has a bug that only allows it to advertise to the first connection.

### Expected Behavior
mdns will advertise all services to all connections in ap mode

### Actual Behavior
mdns only appears to advertise services for first connection in ap mode

### Steps to repropduce

1. use the mdns example in /examples
2. Modify example to run in ap mode (set max connections to some value greater than 1)
3. Connect the ap with one device (you'll see that you can see advertised services using avahi)
4. Connect to the ap with a second device (the second device will not find any services using avahi

I even tried freeing and then re-initing the mdns server for every connection. That did not help with anything.

negativekelvin, 2019-03-31:
Could be related to this:
>Currently ESP32 AP doesn't support all of the power save feature defined in Wi-Fi specification. To be specific, the AP only caches unicast data for the stations connect to this AP, but doesn't cache the multicast data for the stations. If stations connected to the ESP32 AP are power save enabled, they may experience multicast packet loss.

Have you tried with generic udp multicast?  Do all clients receive?

lucascoxBAF, 2019-04-01:
I assume this is unrelated to that. The issues is that only the first station connected to the esp32 softAP sees the mdns advertising. The second station to connect doesn't get the mdns advertisements from the AP. It's not an issues of the esp not caching broadcasts from one device to the other.

negativekelvin, 2019-04-02:
I don't think it applies only station to station

>But because multicast is going to everyone, the AP sends a broadcast to everyone. DTIM does power management but clients can choose to wake up or not and whether to drop the packet or not. Then they don't know why their bonjour doesn't work. 

https://tools.ietf.org/id/draft-mcbride-mboned-wifi-mcast-problem-statement-01.html

lucascoxBAF, 2019-04-02:
Yeah, you’re right. The power-save stations would experience udp loss if the AP didn’t cache the multicasts. 

In the case I reported, though, it’s the second station that never gets any mdns advertisements. It doesn’t just miss some of them, it never gets any of them. If my phone connects first, it gets the broadcasts, and then if my laptop connects, it doesn’t. If the laptop connects first, it gets the mdns messages, and then if the phone connects, it doesn’t.

closed 2019-06-18 by lucascoxBAF