espressif/esp-idf #3240: SPI Slave: Example not work with other devices

mauriciofarina, 2019-04-01:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0-dev-181-ge2ed49c
- Build System:  Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:  Linux
- Power Supply:   USB

## Problem Description

Greetings,

I'm having some trouble implementing a SPI communication between the ESP32 (SLAVE) and the nrf51-dk (MASTER).

Basically, what I'm trying to do is sending a char array from the nrf51 to the ESP32. In order to do so, I'm running a Mbed SPI Master example and trying to receive the array with the esp-idf example (peripherals/spi_slave/receiver).

### Expected Behavior

Receive the char array

### Actual Behavior

Receiving lots of trash.

### Steps to repropduce

On the nrf51 I'm running the code below:

```cpp
#include <mbed.h>

Serial pc(USBTX, USBRX);  //UART
SPI spi(SPI_PSELMOSI0,SPI_PSELMISO0,SPI_PSELSCK0);//SPI
DigitalIn handshake(p23); //SPI Handshake Pin
DigitalOut cs(SPI_PSELSS0); //SPI Chip Select

int main() {

    cs = 1; //Disable Slave

    pc.baud(9600);  //UART Init
    pc.printf("Starting...\n\r");

    spi.format(8,3);  //SPI 8 bits, Mode 3
    spi.frequency(1000000); // 1 MHz clock

    

    while(1) {
       
      //RX TX Buffers
      char tx_buffer[128] = {"MASTER TX BUFFER"};
      char tx_length = 128;
      char rx_buffer[128] = {"MASTER RX BUFFER"};
      char rx_length = 128;
      
      
      cs = 0; //Enable Slave

      while(!handshake){} // Wait for slave to be available

      spi.write(tx_buffer, tx_length, rx_buffer, rx_length);  //Transmission

      cs = 1; // Disable Slave

      pc.printf("String:\n\r"); //Print RX Buffer
      for(i = 0; i < 128 ; i++){
        pc.printf("%c", rx_buffer[i]);
      }
      pc.printf("\n\r");
      
      
      wait(5); // Wait

    }
}

```

On the ESP32 I'm running the SPI_Slave (Receiver) example with a few changes:

- SPI mode 3
- Added the Attribute `WORD_ALIGNED_ATTR` to `recvbuf` and `sendbuf`
- Changed the transmission String

```cpp
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "lwip/igmp.h"

#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/rtc_cntl_reg.h"
#include "esp32/rom/cache.h"
#include "driver/spi_slave.h"
#include "esp_log.h"
#include "esp_spi_flash.h"

#define GPIO_HANDSHAKE 2
#define GPIO_MOSI 12
#define GPIO_MISO 13
#define GPIO_SCLK 15
#define GPIO_CS 14

void my_post_setup_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TS_REG, (1<<GPIO_HANDSHAKE));
}

void my_post_trans_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TC_REG, (1<<GPIO_HANDSHAKE));
}

void app_main()
{
    int n=0;
    esp_err_t ret;

    spi_bus_config_t buscfg={
        .mosi_io_num=GPIO_MOSI,
        .miso_io_num=GPIO_MISO,
        .sclk_io_num=GPIO_SCLK
    };

    spi_slave_interface_config_t slvcfg={
        .mode=3,    // Changed to mode 3
        .spics_io_num=GPIO_CS,
        .queue_size=3,
        .flags=0,
        .post_setup_cb=my_post_setup_cb,
        .post_trans_cb=my_post_trans_cb
    };

    gpio_config_t io_conf={
        .intr_type=GPIO_INTR_DISABLE,
        .mode=GPIO_MODE_OUTPUT,
        .pin_bit_mask=(1<<GPIO_HANDSHAKE)
    };

    gpio_config(&io_conf);

    gpio_set_pull_mode(GPIO_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_SCLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_CS, GPIO_PULLUP_ONLY);

    ret=spi_slave_initialize(HSPI_HOST, &buscfg, &slvcfg, 1);
    assert(ret==ESP_OK);

    WORD_ALIGNED_ATTR char sendbuf[129]=""; // Added the WORD ALIGNED Attribute
    WORD_ALIGNED_ATTR char recvbuf[129]="";
    memset(recvbuf, 0, 33);
    spi_slave_transaction_t t;
    memset(&t, 0, sizeof(t));

    while(1) {
        memset(recvbuf, 0xA5, 129);
        sprintf(sendbuf, "SLAVE TX BUFFER"); // Changed String

        t.length=128*8;
        t.tx_buffer=sendbuf;
        t.rx_buffer=recvbuf;
        
        ret=spi_slave_transmit(HSPI_HOST, &t, portMAX_DELAY);

        printf("Received: %s\n", recvbuf);
        n++;
    }

}
```

[sdkconfig File](https://github.com/espressif/esp-idf/files/3031269/sdkconfig.zip)

Any ideas of what I'm doing wrong?

mauriciofarina, 2019-04-03:
Some updates,

I'd tried connecting the MISO and MOSI pins on each board, and, in both cases, they got able to transmit and receive their own transmissions.

Also, I've notice that the RX buffer from the nrf51 is empty after the transmission is done.

ginkgm, 2019-04-15:
Hi @mauriciofarina ,
can you try to pull down CS wire only after the handshake is active (=0)?

the correct timing is slave transmit->slave pulldown handshake in the callback->master sees the hand shake->master pull low cs->master transmit a few bytes->master pull up cs->slave detect the transmission done->slave pull up handshake

Besides, you may need to wait for some time (I saw you did this in your code, it's great) long enough for the slave to pull high the handshake before the next transmission. Otherwise the master may treat the slave as ready and start a new transmission incorrectly

mauriciofarina, 2019-11-20:
Hello @ginkgm, 

Thanks for the help!
That solved my problem!

closed 2019-11-20 by mauriciofarina