espressif/esp-idf #3241: Touch driver triggers all active channels when touch_pad_trigger_mode is called

Elijahg, 2019-04-01:
## Environment
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3-dev-264-g8a69ffc36
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS 10.14.4
- Power Supply:         USB

## Problem Description

The touch pad driver has strange behaviour when touch interrupts are used with multiple channels, and when touch_pad_set_trigger_mode() is subsequently called. 

If the touch trigger is changed to TOUCH_TRIGGER_ABOVE when some channels are already above their thresholds, the attached ISR will continually fire.  

I would like to detect how long a button has been touched, so I need to detect both when a button is pressed, and when it's released. Do to this, when the ISR is called, I pass the result of touch_pad_get_status() via a message queue to an RTOS task. This then carries out functions related to that button, and crucially, sets the trigger level to slightly above the current value for the pressed button, and sets touch_pad_set_trigger_mode(TOUCH_TRIGGER_ABOVE). This immediately causes the ISR to fire multiple times, as if all the other touch channels have been activated.

### Expected Behavior
The touch channels already above or below the threshold should not trigger the ISR when the trigger mode is changed.

There are several options IMO to resolve this. Obviously some of these may not be possible in hardware only, and would require software fixes. 

-A trigger mode could be set for each channel, this would be a preferred option IMO.
-Only triggering once each time the threshold is passed (similar to GPIO interrupt "falling" or "rising").
-The ability to set the mask (or individual bits within it) returned by touch_pad_get_status(), so that it will not fire again until each bit is cleared.

### Actual Behavior
All ISRs are triggered immediately when touch_pad_set_trigger_mode() is changed.

### Steps to reproduce

Several conditions need to be set for this behaviour to be shown: 
An ISR should be attached, the threshold for several channels set to something less than their current value, interrupts should be enabled and and TOUCH_TRIGGER_BELOW should be the trigger mode. When the touch trigger is then set high, all the ISRs fire.

1) Set touch_pad_set_trigger_mode(TOUCH_TRIGGER_BELOW).
2) Set up multiple touch channels with a threshold below the current non-touched level.
3) Set touch_pad_set_trigger_mode(TOUCH_TRIGGER_ABOVE).
4) All ISRs fire.

A workaround is to disable all other touch channels when a single button is pushed, set the threshold on the channel that is activated, then re-activate the other channels when the button is released. However due to a bug in touch_pad_get_status() that I'll be reporting shortly, detection of a rising button seems unreliable.

### Code to reproduce this issue

[main.cpp](https://gist.github.com/Elijahg/d3c7db79d68ad06c0324fb8c0994975d)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3031219/sdkconfig.txt)

[touch_pad_interrupt.elf.zip](https://github.com/espressif/esp-idf/files/3031221/touch_pad_interrupt.elf.zip)

## Other items if possible

- [√ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ √] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

fuAlex, 2019-04-15:
When the condition is met(TOUCH_TRIGGER_BELOW or TOUCH_TRIGGER_ABOVE), the touch interrupt will continue to be triggered. Just like the level trigger type of GPIO.  
The selection of the trigger mode is global, for all channels.   
Therefore, if you want to test the precise time that the button is continuously touched, you can turn off the enable or increase threshold of other channels in "TRIGGER_BELOW" ISR function, and then modify the trigger mode to "TRIGGER_BELOW", waiting for this button to be released. .

Alvin1Zhang, 2019-05-23:
@Elijahg Hi, Thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2020-05-29:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-05-29 by Alvin1Zhang