espressif/esp-idf #3242: Touch driver's touch_pad_get_status() strange/broken behaviour

Elijahg, 2019-04-02:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3-dev-264-g8a69ffc36
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS 10.14.4
- Power Supply:         USB

## Problem Description
After an interrupt has fired, touch_pad_get_status() function seems to have odd behaviour.

The touch_pad_get_status() function seems to require at least touch channels 0 or 1 to be initialised otherwise it won't return valid values for _any_ channel. Attempting to use only for example channel 2 and 3 results in touch_pad_get_status() returning all zeros, or when **one** of the channels has been touched, 0xC, erroneously indicating that both channel 2 **and** 3 have been touched. This behaviour does not occur if channel 0 or 1 is initialised, and the function seem to return reasonable values.
 
### Expected Behavior
Bits should be set correctly, even if channel 0 or 1 aren't in use

### Actual Behavior
touch_pad_get_status()'s mask indicates that all configured buttons are touched, rather than a mask corresponding to the activated channels.

### Steps to repropduce

1. Set up several touch channels _not including channel 0 or 1_ with ISR attached
2. Set the thresholds to a reasonable level below the untouched level
3. Set the trigger mode to be below
4. Touch a button
5. Print the output of touch_pad_get_status()
6. Observe that the mask of touch_pad_get_status() is incorrect

To verify it's ok with channel 0 or 1 active:
1. Set up several touch channels _including channel 0 or 1_ with ISR attached
2. Set the thresholds to a reasonable level below the untouched level
3. Set the trigger mode to be below
4. Touch a button
5. Print the output of touch_pad_get_status()
6. Observe that the mask of touch_pad_get_status() is correct.

### Code to reproduce this issue

[main.cpp](https://gist.github.com/Elijahg/01099f8b9f1d695a9e0892df61e7142e)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3032107/sdkconfig.txt)
[touch_pad_interrupt.elf.zip](https://github.com/espressif/esp-idf/files/3032108/touch_pad_interrupt.elf.zip)

fuAlex, 2019-04-15:
I have not reproduced your question, the attachment is the code I am using, please check it.
  
[tp_interrupt_main.c.zip](https://github.com/espressif/esp-idf/files/3078901/tp_interrupt_main.c.zip)

Alvin1Zhang, 2019-05-20:
@Elijahg Hi, Thanks for reporting the issue. Sorry that we could not reproduce your issue. Would you help check our codes and help provide more details if possible? Thanks.

Alvin1Zhang, 2020-05-29:
@Elijahg Thanks for reporting, will close due to short of feedback since issue is created, feel free to reopen with more details. Thanks.

closed 2020-05-29 by Alvin1Zhang