espressif/esp-idf #3244: SYSTEM_EVENT_STA_GOT_IP never received, other units work, Wrover-B defective?

lrodenasSistel, 2019-04-02:
Hello,
Please help me debug this issue, right now I believe we got a defective Wrover-B module, but don't know how to confirm this.

At my company, Sistel, we are at the final stage of product development using ESP32 as main MCU. We received this week 5 test units, they all work perfect, except one of them is never getting the IP event so it doesn't have internet connection.

Same HW, same SW, same programming method, etc. Looking at the PCB there is no sign of bad solder or anything different from the other units.

We are using Wrover-B module with 8MB flash and 8MB RAM, PCB antenna, soldered on our pcb which has many other components: RTC crystal, SD card, another MCU, RFID, NFC, Eink, battery charger, buzzer, leds controller. And of course a battery, EINK screen, RFID and NFC antennas. Everything else works correctly in every unit.

IDF release version 3.1.3.

Code is propietary so I cannot publicly share it, yet.

Units that work OK receive events: 
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
SYSTEM_EVENT_STA_GOT_IP 

Unit that is NOT working just receives:
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
nothing else

It does not matter in which order I turn them on, or if the other units are on or off.

My first thought was no wifi communication due to bad antenna matching or something, but then I saw STA connected event, that means it is connecting to the Wifi network, right?

Any clue about what could be happening? Is it feasible that I got 1 defective unit of Wrover-B module from just 5 units? (I find this hard to believe, pretty sure you are testing them on fab right?)
How can I better debug this issue or what extra info may you need in order to better understand what is going on?

Thank you very much for your time and support,
Luis Ródenas

negativekelvin, 2019-04-02:
Turn on debug or verbose logging.  Enable lwip debug logging.

lrodenasSistel, 2019-04-02:
I found the verbose logging option in menuconfig, but how do I enable lwip debug logging?

[image]

negativekelvin, 2019-04-02:
https://github.com/espressif/esp-idf/blob/2f8b6cfc77bdb2c5c412f17529d5fddf9b68b325/components/lwip/port/esp32/include/lwipopts.h#L808

lrodenasSistel, 2019-04-02:
Thank you @negativekelvin !
I changed 
`#define LWIP_DEBUG                      LWIP_DBG_OFF`
to
`#define LWIP_DEBUG                      LWIP_DBG_ON`

And this is the log of the wifi task after enabling LWIP debug and verbose debug.

MONITOR
--- idf_monitor on COM10 2000000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

                                           I (30) boot: ESP-IDF v3.1.3-dirty 2nd stage bootloader
I (30) boot: compile time 13:51:28
I (30) boot: Enabling RNG early entropy source...
I (30) qio_mode: Enabling default flash chip QIO
I (31) boot: SPI Speed      : 80MHz
I (31) boot: SPI Mode       : QIO
I (31) boot: SPI Flash Size : 8MB
I (31) boot: Partition Table:
I (32) boot: ## Label            Usage          Type ST Offset   Length
I (32) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (32) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (33) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (33) boot:  3 ota_0            OTA app          00 10 00010000 002f0000
I (34) boot: End of partition table
I (34) boot: No factory image, trying OTA 0
I (34) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x26e74 (159348) map
I (77) esp_image: segment 1: paddr=0x00036e9c vaddr=0x3ffc0000 size=0x038ac ( 14508) load
I (82) esp_image: segment 2: paddr=0x0003a750 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at C:/DESARROLLO/ESP32/esp-idf/components/freertos/xtensa_vectors.S:1779

I (82) esp_image: segment 3: paddr=0x0003ab58 vaddr=0x40080400 size=0x054b8 ( 21688) load
I (90) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x9831c (623388) map
0x400d0018: _stext at ??:?

I (255) esp_image: segment 5: paddr=0x000d833c vaddr=0x400858b8 size=0x11b28 ( 72488) load
0x400858b8: esp_cpu_in_ocd_debug_mode at C:/DESARROLLO/ESP32/esp-idf/components/soc/esp32/cpu_util.c:58

I (293) boot: Loaded app from partition at offset 0x10000
I (293) boot: Disabling RNG early entropy source...
I (294) spiram: SPI RAM mode: flash 80m sram 80m
I (294) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (294) cpu_start: Pro cpu up.
I (294) cpu_start: Starting app cpu, entry point is 0x4008140c
0x4008140c: call_start_cpu1 at C:/DESARROLLO/ESP32/esp-idf/components/esp32/cpu_start.c:232

I (0) cpu_start: App cpu up.
I (791) spiram: SPI SRAM memory test OK
I (791) heap_init: Initializing. RAM available for dynamic allocation:
D (791) heap_init: New heap initialised at 0x3ffaff10
I (791) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (792) heap_init: New heap initialised at 0x3ffca518
I (792) heap_init: At 3FFCA518 len 00015AE8 (86 KiB): DRAM
I (792) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (793) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (793) heap_init: New heap initialised at 0x400973e0
I (793) heap_init: At 400973E0 len 00008C20 (35 KiB): IRAM
I (794) cpu_start: Pro cpu start user code
I (794) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (794) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (795) clk: waiting for 32k oscillator to start up
D (858) clk: RTC_SLOW_CLK calibration value: 16000000
V (187) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (188) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (188) intr_alloc: Connected src 46 to int 2 (cpu 0)
V (188) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (188) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC0E
D (189) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (189) esp_dbg_stubs: esp_dbg_stubs_init stubs 3ffc4108
[... 143 lines cut ...]
D (1312) nvs: nvs_set_blob sta.apinfo 700
I (1312) wifi: pm start, type: 0

D (1312) event: SYSTEM_EVENT_STA_CONNECTED, ssid:SistelGUEST, ssid_len:11, bssid:82:2a:a8:c7:fe:09, channel:1, authmode:3
V (1312) event: enter default callback
V (1312) tcpip_adapter: check: local, if=0 fn=0x400fe620
0x400fe620: tcpip_adapter_up_api at C:/DESARROLLO/ESP32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1089

V (1312) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (1322) tcpip_adapter: check: remote, if=0 fn=0x400fe620
0x400fe620: tcpip_adapter_up_api at C:/DESARROLLO/ESP32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1089

V (1322) tcpip_adapter: check: local, if=0 fn=0x400feb0c
0x400feb0c: tcpip_adapter_dhcpc_start_api at C:/DESARROLLO/ESP32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1089

D (1322) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (1322) tcpip_adapter: if0 start ip lost tmr: enter
D (1322) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffe3bf8 interval=120 ip=0
D (1322) tcpip_adapter: dhcp client start successfully
V (1322) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (1322) tcpip_adapter: check: remote, if=0 fn=0x400feb0c
0x400feb0c: tcpip_adapter_dhcpc_start_api at C:/DESARROLLO/ESP32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1089

V (1322) event: exit default callback
EVENT STA CONNECTED 4

lrodenasSistel, 2019-04-02:
Nothing else happens after this, no more events received, and the task is blocked waiting for IP event.

I am using IDF v3.1.3 (my lwip file was a bit different that the one referenced I guess because of IDF version), and my sdkconfig is the following:

```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="COM10"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=y
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=2000000
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=y
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHFREQ_40M=
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="80m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="8MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
[... 725 lines cut ...]
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

```

lrodenasSistel, 2019-04-03:
Does anyone else see anything weird in this log that could prevent this unit from getting IP event?
Or please let me know if I can provide any extra information.

Any help or hint is much appreciated. Thank you

lrodenasSistel, 2019-04-08:
We found out the problem!!!
After some investigation we found out our DHCP server in my company was working erratically, not giving IP to some specific MAC addresses, but working ok with the vast majority. No idea why. We have Windows Server 2012, we removed the option of DHCP redundancy and restarted the service and everything seems to be working OK again, with all MAC addresses.

Conclusion: if you are not getting IP event, DHCP in the wifi network may be the cause.

You may close this case now, thank you.

closed 2019-04-08 by lrodenasSistel