espressif/esp-idf #3245: "hwcrypto/sha.h" library either depriciate or remove

manish5897, 2019-04-02:
I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.

Now, might be the library is removed; It seems fatal error: hwcrypto/sha.h: No such file or directory
compilation terminated.

Please let me know the alternate for the same.

suda-morris, 2019-04-02:
Sorry about that. 
We've move the original hwcrypto/sha.h to https://github.com/espressif/esp-idf/tree/master/components/mbedtls/port/include/esp32/sha.h. So in your application where uses `include hwcrypto/sha.h` should be replaced by `include esp32/sha.h` now.
We will add fallback as soon as possible.

projectgus, 2019-04-03:
Hi @manish5897 ,

Alternatively, you can include "mbedtls/sha.h" and rename the esp32_xxx APIs to mbedtls_xxx. This will work in both IDF v3.x and v4.x, and [uses hardware acceleration if configured in menuconfig](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/kconfig.html#envvar-CONFIG_MBEDTLS_HARDWARE_SHA). There is no performance overhead to calling the mbedTLS APIs instead of the "esp32" APIs.

closed 2019-07-24 by manish5897