espressif/esp-idf #3246: Wrong reason for wrong password during connection in STA (Wifi)

jmberhault, 2019-04-02:
## Environment

- Module or chip used:  ESP32-WROOM-32 & ESP32-WROOM-32D
- IDF version :  v3.2-beta3-72-g70811110f
- Build System: Make
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- Power Supply: USB & external 3.3V

## Problem Description

Wrong reason code in SYSTEM_EVENT_STA_DISCONNECTED event when the pw is wrong.

### Expected Behavior

Getting the correct reason codes in the event in case of the wrong password => I'm expecting this error : WIFI_REASON_AUTH_FAIL (202) ??

### Actual Behavior

Return error code 2 (WIFI_REASON_AUTH_EXPIRE) and error code 201 (WIFI_REASON_NO_AP_FOUND)

### Steps to reproduce

Try to connect with a wrong password my local hotspot (of course on).
Others functions are working well, wps with push-button is working too.

### Code to reproduce this issue

To reproduce it, I've taken the example wifi/station where I have changed in the define the SSID and the password (theses two lines only). I have modified the debug mode for more logs as below.

## Debug Logs

### working logs - (only the password have been replaced by xxxxxxxx)
`
D (345) nvs: nvs_set sta.sort_method 1 0[1B][0m
D (345) nvs: nvs_set sta.minrssi 1 -127[1B][0m
D (355) nvs: nvs_set sta.minauth 1 0[1B][0m
D (355) nvs: nvs_set_blob sta.apsw 2[1B][0m
D (355) nvs: nvs_set_blob sta.apinfo 700[1B][0m
D (365) RTC_MODULE: Wi-Fi takes adc2 lock.[1B][0m
D (365) phy_init: loading PHY init data from application binary[1B][0m
D (375) nvs: nvs_open_from_partition phy 0[1B][0m
D (375) nvs: nvs_get cal_version 4[1B][0m
D (375) nvs: nvs_get_str_or_blob cal_mac[1B][0m
D (385) nvs: nvs_get_str_or_blob cal_data[1B][0m
D (395) nvs: nvs_close 3[1B][0m
[1B][0;32mI (445) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0[1B][0m
I (445) wifi: mode : sta (30:ae:a4:80:17:60)
D (445) event: SYSTEM_EVENT_STA_START[1B][0m
[1B][0;32mI (455) wifi station: wifi_init_sta finished.[1B][0m
[1B][0;32mI (455) wifi station: connect to ap SSID:Livebox-8656 password:_xxxxxxxxxx_[1B][0m
I (575) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1555) wifi: state: init -> auth (b0)
I (1555) wifi: state: auth -> assoc (0)
I (1565) wifi: state: assoc -> run (10)
I (1575) wifi: connected with Livebox-8656, channel 1
D (1575) nvs: nvs_set sta.chan 1 1[1B][0m
D (1585) nvs: nvs_set_blob sta.apinfo 700[1B][0m
I (1585) wifi: pm start, type: 1

D (1585) event: SYSTEM_EVENT_STA_CONNECTED, ssid:Livebox-8656, ssid_len:12, bssid:34:8a:ae:24:86:56, channel:1, authmode:3[1B][0m
D (1595) tcpip_adapter: dhcp client init ip/mask/gw to all-0[1B][0m
D (1605) tcpip_adapter: if0 start ip lost tmr: enter[1B][0m
D (1605) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc5860 interval=120 ip=0[1B][0m
D (1615) tcpip_adapter: dhcp client start successfully[1B][0m
D (1755) phy_init: wifi mac time delta: 32571[1B][0m
D (2055) phy_init: wifi mac time delta: 215682[1B][0m
D (2365) phy_init: wifi mac time delta: 164039[1B][0m
D (2655) phy_init: wifi mac time delta: 175625[1B][0m
D (2655) tcpip_adapter: if0 dhcpc cb[1B][0m
[... 122 lines cut ...]
D (18825) tcpip_adapter: if0 start ip lost tmr: enter[1B][0m
D (18835) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc58e4 interval=120 ip=0[1B][0m
[1B][0;32mI (18845) wifi station: retry to connect to the AP[1B][0m
[1B][0;32mI (18845) wifi station: connect to the AP fail
[1B][0m
D (21255) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Livebox-8656, ssid_len:12, bssid:00:00:00:00:00:00, reason:201[1B][0m
D (21255) tcpip_adapter: if0 start ip lost tmr: enter[1B][0m
D (21265) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc58e4 interval=120 ip=0[1B][0m
[1B][0;32mI (21275) wifi station: connect to the AP fail

## Other items if possible

My hotspot is a french livebox from Orange : livebox 3 FR manufactured by Sagemcom
firmware revision : SoftAtHome SG30_sip-fr-6.52.12.1
orange firmware revision : g6-f-sip-fr
SSID  : Livebox-8656
channel  : 1
standard : 802.11bgn
maximum speed : 450 Mb/s
security : WPA/WPA2 Mixed
security option : TKIP+AES

I'm working for few projets with two types of EPS32 (4 or 8Mb - WROOM-32 and WROOM-32D), and it's the same result, also with example as you could see here.

Please let me know if you need more logs, more information, more tests, to resolve this issue.

liuzfesp, 2019-04-02:
Hi @jmberhault we already have fixed  for this issue, I think the fix will be backport to v3.0/v3.1/v3.2/v3.3 soon.

crc1021, 2019-04-02:
this might be related, I just noticed WPS is not working v3.3 branch, but it is working v3.1 and v3.2
In v.3.3, the log shows continual  SYSTEM_EVENT_STA_DISCONNECTED using the stock example in examples/wifi/wps 
```
I (349) wifi: mode : sta (30:ae:a4:d4:23:2c)
I (349) example_wps: start wps...
I (349) example_wps: SYSTEM_EVENT_STA_START
I (359) wpa: manufacturer: ESPRESSIF, model number: ESP32, model name: ESPRESSIF IOT, device name: ESP STATION
I (369) wpa: wifi_wps_enable

I (5789) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (7839) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
```

liuzfesp, 2019-04-04:
HI @crc1021, could you help to provide the IDF commit ID and wifi firmware ID?

crc1021, 2019-04-04:
for idf
7b368fc2110f7f6dcb8f6ce154c07f8a5db45c4d refs/heads/release/v3.3
for wifi firmware, do you want the commit for esp32/lib submodule?
064ebf1b04bc9eed3b038fec0344d344ed05e60b components/esp32/lib

liuzfesp, 2019-04-23:
Hi @crc1021, this issue has been fixed in IDF master: ebe7190f, please help to check it.

crc1021, 2019-04-23:
the problem I had was not due to this, I filed a separate issue #3294 for problem with WPS; #3318 appears to be same problem

closed 2019-11-07 by Alvin1Zhang