espressif/esp-idf #3247: Loading takes a very long time.

ArTyGill, 2019-04-02:
I want to use an external watchdog timer that monitors the pin esp32, but the download lasts 3 seconds and the external watchdog timer is triggered before the controller starts.
[image]

ArTyGill, 2019-04-02:
20:25:13.839> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:25:13.839> configsip: 0, SPIWP:0xee
20:25:13.839> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:25:13.839> mode:FAST_RD, clock div:1
20:25:13.839> load:0x3fff0018,len:4
20:25:13.839> load:0x3fff001c,len:808
20:25:13.839> load:0x40078000,len:6084
20:25:13.839> load:0x40080000,len:6696
20:25:13.839> entry 0x400802e4
20:25:15.715> ets Jun  8 2016 00:22:57
20:25:15.715> 
20:25:15.715> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:25:15.715> configsip: 0, SPIWP:0xee
20:25:15.715> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:25:15.715> mode:FAST_RD, clock div:1
20:25:15.715> load:0x3fff0018,len:4
20:25:15.715> load:0x3fff001c,len:808
20:25:15.715> load:0x40078000,len:6084
20:25:15.715> load:0x40080000,len:6696
20:25:15.715> entry 0x400802e4
20:25:17.528> ets Jun  8 2016 00:22:57
20:25:17.528> 
20:25:17.528> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:25:17.528> configsip: 0, SPIWP:0xee
20:25:17.528> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:25:17.528> mode:FAST_RD, clock div:1
20:25:17.528> load:0x3fff0018,len:4
20:25:17.528> load:0x3fff001c,len:808
20:25:17.528> load:0x40078000,len:6084
20:25:17.528> load:0x40080000,len:6696
20:25:17.528> entry 0x400802e4
20:25:19.402> ets Jun  8 2016 00:22:57
20:25:19.402> 
20:25:19.402> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:25:19.402> configsip: 0, SPIWP:0xee
20:25:19.402> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:25:19.402> mode:FAST_RD, clock div:1
20:25:19.402> load:0x3fff0018,len:4
20:25:19.402> load:0x3fff001c,len:808
20:25:19.402> load:0x40078000,len:6084
20:25:19.402> load:0x40080000,len:6696
20:25:19.402> entry 0x400802e4
20:25:21.279> ets Jun  8 2016 00:22:57
20:25:21.279> 
20:25:21.279> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:25:21.279> configsip: 0, SPIWP:0xee
20:25:21.279> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:25:21.279> mode:FAST_RD, clock div:1
20:25:21.279> load:0x3fff0018,len:4
20:25:21.279> load:0x3fff001c,len:808
20:25:21.279> load:0x40078000,len:6084
20:25:21.279> load:0x40080000,len:6696
20:25:21.279> entry 0x400802e4

ArTyGill, 2019-04-02:
The external watchdog timer is activated after 2 seconds, and the esp32 needs about 3 seconds to load.

Elijahg, 2019-04-03:
Why don't you use the internal watchdog?

projectgus, 2019-04-03:
Hi @ArTyGill ,

There are some things you can do to speed up booting:

* You have the clock set to `clock div: 1` which is good (maximum SPI flash clock speed)
* For some reason your flash read mode is `FAST_RD`, this is the slowest mode. Can you [configure your firmware for QIO mode](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/kconfig.html#envvar-CONFIG_FLASHMODE), and when flashing the bootloader specify `DIO` mode ? (bootloader can't do QIO itself)
* You seem to have logging configured in the bootloader & app at a lower level (Errors only or disabled), which is good as it will speed up the initial boot. You can also disable all boot messages from the ROM stage by pulling MTDO (GPIO15) low, although this might only save a few milliseconds.
* If you're not feeding the watchdog as the first thing your firmware does in app_main(), you may want to change that.
* This should get boot time under 2 seconds. If it doesn't, you can patch the bootloader (make a copy of `IDF_PATH/components/bootloader` under your own project's components directory) and edit `bootloader/subproject/main/bootloader_start.c` to "feed" the watchdog at different stages of the bootloader process. You can include "rom/gpio.h" to access some ROM GPIO functions.

ArTyGill, 2019-04-03:
@Elijahg I use an internal watchdog but he
not very reliable. In some cases it may hang.

Elijahg, 2019-04-03:
That's strange, I've never had the watchdog hang myself. Perhaps your power supply is noisy, maybe lacking decoupling capacitors? Seems like Projectgus has provided plenty of ways to fix it anyway!

ArTyGill, 2019-04-11:
@Elijahg @projectgus  Yes, thanks, for the help of the only option I see to ask the producer's supervisor for a different delay time.

ArTyGill, 2019-04-11:
@Elijahg This is not related to power, I used arduino, and on the branch 1.0.1 a week later the controller was hanging tight.

projectgus, 2019-04-11:
@ArTyGill Did you try switching from FAST_RD mode to a different bootloader mode, as suggested? This will make a big difference to boot time.

If the question you had is resolved, please close this issue.

closed 2019-04-12 by ArTyGill