espressif/esp-idf #3251: esp_ota_get_app_description missing

tatulea, 2019-04-03:
## Environment

- IDF version (run ``git describe --tags`` to find it): v3.1.3
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:    Linux

## Problem Description

I am trying to get the app version using the esp_ota_get_app_description, but the problem is that the function does not exists. It should be defined in components/app_updates, but these files seems to be older and they are not the same version as the one from git. I don't even have all the files from components/app_updates. What should I do?

chegewara, 2019-04-03:
https://esp32.com/viewtopic.php?f=13&t=9950#p41073

tatulea, 2019-04-04:
Thanks! The means that we have to wait for V3.3

closed 2019-04-04 by tatulea