espressif/esp-idf #3252: The provided example "/example/wifi/scan" is wrong

GitOfLance, 2019-04-03:
### Environment

- Module or chip used: ESP32-WROOM-32 
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
Use demo of [examples/wifi/scan](https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan) can't get AP information. Because **esp_wifi_scan_start**() is not called in the source code and when the scan-done event arise, the function **esp_wifi_scan_get_ap_num()**、**esp_wifi_scan_get_ap_records()** are not called to do user-defined things. Of course, no sacn-done event arise because no scan is started.

### Debug Logs
```
I (0) cpu_start: App cpu up.
I (371) heap_init: Initializing. RAM available for dynamic allocation:
I (378) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (384) heap_init: At 3FFB8DC0 len 00027240 (156 KiB): DRAM
I (390) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (397) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (403) heap_init: At 4008FB50 len 000104B0 (65 KiB): IRAM
I (409) cpu_start: Pro cpu start user code
I (92) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (176) wifi: wifi driver task: 3ffc0ae0, prio:23, stack:3584, core=0
I (176) wifi: wifi firmware version: 8b94b0e
I (176) wifi: config NVS flash: enabled
I (176) wifi: config nano formating: disabled
I (186) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (186) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (226) wifi: Init dynamic tx buffer num: 32
I (226) wifi: Init data frame dynamic rx buffer num: 32
I (226) wifi: Init management frame dynamic rx buffer num: 32
I (226) wifi: Init management short buffer num: 32
I (236) wifi: Init static rx buffer size: 1600
I (236) wifi: Init static rx buffer num: 10
I (246) wifi: Init dynamic rx buffer num: 32
I (356) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (356) wifi: mode : sta (30:ae:a4:15:d0:ac)
I (356) scan: SYSTEM_EVENT_STA_START
I (2416) scan: SYSTEM_EVENT_STA_DISCONNECTED
I (4466) scan: SYSTEM_EVENT_STA_DISCONNECTED
I (6516) scan: SYSTEM_EVENT_STA_DISCONNECTED
I (8576) scan: SYSTEM_EVENT_STA_DISCONNECTED
```

### Improved code
Refer to the official instructions [Wi-Fi Driver](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-guides/wifi.html?highlight=esp_wifi_scan_start#esp32-wi-fi-feature-list), the code that can be executed correctly after modification.
```
/* Scan Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
    This example shows how to use the All Channel Scan or Fast Scan to connect
    to a Wi-Fi network.

    In the Fast Scan mode, the scan will stop as soon as the first network matching
    the SSID is found. In this mode, an application can set threshold for the
    authentication mode and the Signal strength. Networks that do not meet the
    threshold requirements will be ignored.

    In the All Channel Scan mode, the scan will end only after all the channels
    are scanned, and connection will start with the best network. The networks
    can be sorted based on Authentication Mode or Signal Strength. The priority
[... 145 lines cut ...]
                  codemao6    |     -80    |        WIFI_AUTH_WPA2_PSK
             ChinaNet-WMCJ    |     -81    |    WIFI_AUTH_WPA_WPA2_PSK
             ChinaNet-PVvU    |     -84    |    WIFI_AUTH_WPA_WPA2_PSK
                      ZGJJ    |     -89    |    WIFI_AUTH_WPA_WPA2_PSK
             ChinaNet-2J4u    |     -91    |    WIFI_AUTH_WPA_WPA2_PSK
                  codemao7    |     -93    |        WIFI_AUTH_WPA2_PSK
I (5986) scan: SYSTEM_EVENT_SCAN_DONE
--------scan count of AP is 12-------
======================================================================
             SSID             |    RSSI    |           AUTH           
======================================================================
                  ai-alloy    |     -68    |    WIFI_AUTH_WPA_WPA2_PSK
                  codemao5    |     -71    |        WIFI_AUTH_WPA2_PSK
                ai-alloy-1    |     -73    |        WIFI_AUTH_WPA2_PSK
                  codemao5    |     -76    |        WIFI_AUTH_WPA2_PSK
                  codemao6    |     -82    |        WIFI_AUTH_WPA2_PSK
             ChinaNet-PVvU    |     -84    |    WIFI_AUTH_WPA_WPA2_PSK
                      ZGJJ    |     -89    |    WIFI_AUTH_WPA_WPA2_PSK
               66666666666    |     -93    |    WIFI_AUTH_WPA_WPA2_PSK
                 eucleia08    |     -94    |    WIFI_AUTH_WPA_WPA2_PSK
         vYou_DDPai_M6Plus    |     -95    |        WIFI_AUTH_WPA2_PSK
                    E-test    |     -95    |    WIFI_AUTH_WPA_WPA2_PSK
                  codemao7    |     -96    |        WIFI_AUTH_WPA2_PSK

```

GitOfLance, 2019-04-03:
I see，the example is [Scan for a Specific AP in All Channels](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-guides/wifi.html#scan-for-a-specific-ap-in-all-channels). But this situation is more like esp wifi connect.

Alvin1Zhang, 2019-04-04:
@GitOfLance Thanks for reporting the issue, we will look into this. Thanks.

liuzfesp, 2019-04-04:
HI @GitOfLance, thanks for your suggestion, we will improve it. @espxiehang please help to improve it accordingly.

GitOfLance, 2019-04-05:
@Alvin1Zhang @liuzfesp It's my pleasure. I am honored to provide a little help to reduce the confusion of novices.

bbulkow, 2020-01-05:
I was looking for this same example and found the scan still checked into the codebase at dec 2019 has the ineffective scan code. Thanks Lance for publishing a much better example here, I will steal it, I look forward to the better scan example in a future release - perhaps it can make 4.0?

bbulkow, 2020-01-05:
One additional comment. This code is different from the wifi examples provided in esp-idf because the event loops are different - default vs normal. The documentation from ESP says that wifi events require a default event loop ( "default" meaning "system" in this case, where the "regular" event loop is "user" ). I'm not quite sure how to square the documentation vs the fact that this example works just peachy, but it does.

If ESP could clarify in the core documentation the difference between these event loops, and whether default is actually required, that would be awesome. I hate to do something undocumented but that works.

( My project right now is creating a library for ESP-IDF similar to the much loved "WifiMulti" library which everyone uses for Arduino ESP development. Will publish if it comes to fruition. )

sagb2015, 2020-01-08:
Hi @bbulkow, 
Seems like scan changes are not backported to 4.0. The refactoring was done in 504172e8. We will backport to 4.0 release soon.

Alvin1Zhang, 2020-05-29:
@GitOfLance Thanks for reporting and @bbulkow for sharing updates, and sorry for slow turnaround. The fix on release/4.0 has been available at https://github.com/espressif/esp-idf/commit/2c90bce9807520e0eea056ad9361a6f6e4053540, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-29 by Alvin1Zhang