espressif/esp-idf #3253: Wifi Crash on init wDev_Rxbuf_Init

sebirdman, 2019-04-03:
## Environment
- Development Kit: [none]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version: f49e2afe946f414e1a9f2d7426af6665eab3731b
- Development Env: [vscode]
- Operating System: [Ubuntu]
- Power Supply: [external 5V]

## Problem Description

I've received a few coredumps that have stacks that seem to imply that wifi is crashing:
```

#0  0x4015d628 in wDev_Rxbuf_Init ()
#1  0x401551bb in ic_init ()
#2  0x40146f5e in chip_post_init ()
#3  0x40147111 in wifi_init_process ()
#4  0x4014823a in ieee80211_ioctl_process ()
#5  0x4008ed77 in ppTask ()
#6  0x4009366c in vPortTaskWrapper (pxCode=0x4008ecb8 <ppTask>, pvParameters=0x0) at .../idf/components/freertos/port.c:143
```

The device seems to recover - enough to upload this coredump to my  s3 bucket at least.

## Debug Logs

I've got several other coredumps with this data. let me know if more info is helpful. I can also provide the elf privately if that helps.

liuzfesp, 2019-06-06:
HI @sebirdman, sorry for late response for this issue. Do you still have the *elf file that generates this crash? Is this issue easy to reproduced?

liuzfesp, 2019-11-05:
@sebirdman we just can't reproduce this issue with our stress test, could you provide detailed steps to reproduce this issue. Is this issue easy reproduced on your side?

liuzfesp, 2019-11-14:
Hi @sebirdman, does this issue still exist?

Alvin1Zhang, 2019-11-14:
@sebirdman Thanks for reporting. Will close the ticket due to short of feedback. Feel free to reopen with more details. Thanks.

closed 2019-11-14 by Alvin1Zhang