espressif/esp-idf #3254: ESP_MQTT causes WatchDog Trigger after 3.2-RC Update

sysizlayan, 2019-04-04:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v3.2-rc
- Build System:         [Make|CMake]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:    Linux
- Power Supply:         USB

## Problem Description
We have updated the ESP-IDF to latest release(V3.2-RC). The mqtt_task causes task_wdt to trigger. 
### Steps to repropduce

1. We were using latest branch of the MQTT.
2. Updated the ESP-IDF to V3.2-RC release
3. Started the device in STA mode and connected the broker with TLS.
4. Subscribed to a topic, received the retained message.
5. After 1 minute, mqtt_task running on Core 0 caused the task_wdt to trigger.

[image]

ESP-MQTT configuration from sdkconfig:
#
# ESP-MQTT Configurations
#
CONFIG_MQTT_PROTOCOL_311=y
CONFIG_MQTT_TRANSPORT_SSL=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET=
CONFIG_MQTT_USE_CUSTOM_CONFIG=
CONFIG_MQTT_TASK_CORE_SELECTION_ENABLED=y
CONFIG_MQTT_USE_CORE_0=y
CONFIG_MQTT_USE_CORE_1=
CONFIG_MQTT_CUSTOM_OUTBOX=

Core Dump:
Backtrace: 0x400958dc:0x3ffb05b0 0x40095b91:0x3ffb05d0 0x400d2def:0x3ffb05f0 0x400854ed:0x3ffb0610 0x4000bfed:0x3fff3df0 0x40092c65:0x3fff3e00 0x400917b1:0x3fff3e20 0x4008625d:0x3fff3e60 0x40086313:0x3fff3e90 0x400ed485:0x3fff3eb0 0x4015ff1d:0x3fff3f00 0x4016426a:0x3fff3f20 0x4018a651:0x3fff3f50 0x40164342:0x3fff3f70 0x4018a615:0x3fff3f90 0x4015979c:0x3fff3fb0 0x40159acc:0x3fff3fe0 0x40091d5d:0x3fff4000
0x400958dc: invoke_abort at /home/safa/esp/esp-idf/components/esp32/panic.c:707

0x40095b91: abort at /home/safa/esp/esp-idf/components/esp32/panic.c:707

0x400d2def: task_wdt_isr at /home/safa/esp/esp-idf/components/esp32/task_wdt.c:252

0x400854ed: _xt_lowint1 at /home/safa/esp/esp-idf/components/freertos/xtensa_vectors.S:1154

0x40092c65: vTaskExitCritical at /home/safa/esp/esp-idf/components/freertos/tasks.c:5093

0x400917b1: xQueueGenericSend at /home/safa/esp/esp-idf/components/freertos/queue.c:2038

0x4008625d: lock_release_generic at /home/safa/esp/esp-idf/components/newlib/locks.c:207

0x40086313: _lock_release at /home/safa/esp/esp-idf/components/newlib/locks.c:213

0x400ed485: esp_vfs_select at /home/safa/esp/esp-idf/components/vfs/vfs.c:827

0x4015ff1d: select at /home/safa/esp/esp-idf/components/newlib/select.c:62

0x4016426a: ssl_poll_read at /home/safa/esp/esp-idf/components/tcp_transport/transport_ssl.c:89

0x4018a651: esp_transport_poll_read at /home/safa/esp/esp-idf/components/tcp_transport/transport.c:184

0x40164342: ssl_read at /home/safa/esp/esp-idf/components/tcp_transport/transport_ssl.c:125

0x4018a615: esp_transport_read at /home/safa/esp/esp-idf/components/tcp_transport/transport.c:168

0x4015979c: mqtt_process_receive at /home/safa/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:951

0x40159acc: esp_mqtt_task at /home/safa/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:951

0x40091d5d: vPortTaskWrapper at /home/safa/esp/esp-idf/components/freertos/port.c:435

Thanks in advance.

david-cermak, 2019-04-08:
Hi @sysizlayan , 
can you please repeat your test with enabled `Support LWIP socket select() only` (you can find this option in menuconfig->components->lwip) ?

Thanks

sysizlayan, 2019-04-17:
Thanks for the response @david-cermak 
We have identified the problem. During development, there were multiple devices which are subscribed to the same broker with the same client name.
I cannot explain why it causes watchdog trigger but when the client names are changed, the problem resolved.
Do you have any comment on this issue? If so please share. If not, you can close the issue.

Thanks again :)

david-cermak, 2019-04-24:
Thanks @sysizlayan  for sharing your results. 
Please keep this open, I would like to test the scenario with multiple clients with the same `client_id` to see if some deadlocks or any other issues were not introduced. (Per initial testing it seems to behave as expected, once a new client with the same id connects the other one gets disconnected)

david-cermak, 2020-01-17:
@sysizlayan , sorry for the slow turnaround. wanted to test thoroughly the scenario of multiple mqtt-clients with the same id connecting to one server and forgot to update the issue here. Everything works as expected, once another client connects the previously connected one gets disconnected and this repeats with the reconnection periods. We can close this issue then.

Alvin1Zhang, 2020-01-17:
@sysizlayan Thanks for reporting. Feel free to reopen. Thanks.

closed 2020-01-17 by Alvin1Zhang