espressif/esp-idf #3255: nvs_erase_key and nvs_erase_all work not correctly

k-shyshkin, 2019-04-04:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v2
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0-dev-225-g2f8b6cfc7
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:  Linux
- Power Supply: USB

## Problem Description

`nvs_erase_key` and `nvs_erase_all` functions work not correctly. To delete page or key need firstly try to read one time after, and only after, key or page will be deleted   

### Expected Behavior

After call `nvs_commit` function, next time when i try read deleted key or open deleted page i must get  `ESP_ERR_NVS_NOT_FOUND` error

### Actual Behavior
#### nvs_erase_key
If after `nvs_erase_key` and `nvs_commit` i try get length of data by call `nvs_get_blob(handle, KEY, NULL, &len)` i get `ESP_OK` and real len of erased data.
#### nvs_erase_all
If after `nvs_erase_all` and `nvs_commit` i try open with `NVS_READONLY` flag deleted page, i receive `ESP_OK`

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include <string.h> 
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "esp_log.h"

#define TAG "NVS_EXAMPLE"
#define NVS_PAGE_KEY "storage"
#define NVS_KEY_DATA "data"

void app_main()
{
    size_t len_data = sizeof(uint32_t) * 2;
    uint32_t some_data[len_data];
    uint32_t getting_some_data[len_data];

    some_data[0] = 42;
    some_data[1] = 64;
    // Initialize NVS
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    // Open NVS
    nvs_handle my_handle;
    err = nvs_open(NVS_PAGE_KEY, NVS_READWRITE, &my_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Error (%s) opening NVS handle!", esp_err_to_name(err));
    }
    else
    {
        ESP_LOGI(TAG, "nvs_open success");
        //set some data
        err = nvs_set_blob(my_handle, NVS_KEY_DATA, (void*)some_data, len_data);
[... 196 lines cut ...]
I (258) heap_init: At 4008906C len 00016F94 (91 KiB): IRAM
I (264) cpu_start: Pro cpu start user code
I (282) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (365) NVS_EXAMPLE: nvs_open success
I (365) NVS_EXAMPLE: nvs_set_blob success
I (365) NVS_EXAMPLE: nvs_commit success
I (365) NVS_EXAMPLE: nvs_open success
I (375) NVS_EXAMPLE: some_data nvs_get_blob success
source some_data[0] = 42
 source some_data[1] = 64
getting_some_data[0] = 42
getting_some_data[1] = 64
I (385) NVS_EXAMPLE: nvs_open success
I (395) NVS_EXAMPLE: nvs_erase_key success
I (395) NVS_EXAMPLE: nvs_commit success
I (395) NVS_EXAMPLE: nvs_open success
E (405) NVS_EXAMPLE: Error (ESP_OK) NVS_KEY_DATA still available, but data clean. len_data = 8 
I (415) NVS_EXAMPLE: Error (ESP_ERR_NVS_NOT_FOUND) Now all good, no NVS_KEY_DATA in NVS
I (425) NVS_EXAMPLE: nvs_open success
I (435) NVS_EXAMPLE: nvs_erase_all success
I (435) NVS_EXAMPLE: nvs_commit success
E (435) NVS_EXAMPLE: Error (ESP_OK) NVS page still available, but clean

```

negativekelvin, 2019-04-05:
When you say page, you must mean namespace because multiple namespaces can share the same physical page within a partition.  You suggest nvs_erase_all should erase/unregister the namespace, not just clear the keys/values.

k-shyshkin, 2019-04-05:
@negativekelvin Yes, you are right. `nvs_erase_all` clear the keys/values and erase/unregister the namespace. Did i wrong?

k-shyshkin, 2019-04-05:
@negativekelvin I think about `nvs_erase_all` may work without erase/unregister the namespace (Although it does not make sense, since the namespace will be created as soon as we try to open it in mode `NVS_READWRITE`), but about `nvs_erase_key` : i think `nvs_get_<str/blob>` must not return last size of data, after `nvs_erase_key` calling, when i try request len of data. I think after calling `nvs_erase_key` must return `ESP_ERR_NVS_NOT_FOUND` as in my example when i try call `nvs_get_blob` second time after `nvs_erase_key` calling.

negativekelvin, 2019-04-05:
Yes, nvs_erase_key is not working properly

```
I (377) NVS_EXAMPLE: nvs_open success
I (387) NVS_EXAMPLE: nvs_set_blob success
I (387) NVS_EXAMPLE: nvs_commit success
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=92 used=44 erased=48
 89: W ns= 4 type=66 span=  2 key="data" chunkIdx=0 len=8
 90: D
 91: W ns= 4 type=72 span=  1 key="data" chunkIdx=255 len=-1
I (857) NVS_EXAMPLE: nvs_open success
I (857) NVS_EXAMPLE: some_data nvs_get_blob success
source some_data[0] = 42
 source some_data[1] = 64
getting_some_data[0] = 42
getting_some_data[1] = 64
I (877) NVS_EXAMPLE: nvs_open success
I (887) NVS_EXAMPLE: nvs_erase_key success
I (887) NVS_EXAMPLE: nvs_commit success
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=92 used=42 erased=50
 89: X
 90: X
 91: W ns= 4 type=72 span=  1 key="data" chunkIdx=255 len=-1
I (1357) NVS_EXAMPLE: nvs_open success
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=92 used=42 erased=50
 89: X
 90: X
 91: W ns= 4 type=72 span=  1 key="data" chunkIdx=255 len=-1
 E (1837) NVS_EXAMPLE: Error (ESP_OK) NVS_KEY_DATA still available, but data clean. len_data = 8
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=92 used=42 erased=50
 89: X
 90: X
 91: W ns= 4 type=72 span=  1 key="data" chunkIdx=255 len=-1
I (2317) NVS_EXAMPLE: Error (ESP_ERR_NVS_NOT_FOUND) Now all good, no NVS_KEY_DATA in NVS
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=92 used=41 erased=51
 89: X
 90: X
 91: X
```

negativekelvin, 2019-04-05:
@igrr I think this is because nvs_erase_key will call Storage::eraseItem with ItemType::ANY and eraseMultiPageBlob will not be called?

https://github.com/espressif/esp-idf/blob/a4aadd19267c210690819d936599471c8903aea9/components/nvs_flash/src/nvs_storage.cpp#L520

sagb2015, 2019-04-06:
@negativekelvin:  Yes your reasoning is correct. We found this issue during our internal testing and the fix is already waiting to be merged and should be available soon.  The initial assumption was erasing any of the blob chunk/index will be sufficient to return NOT_FOUND error when blob is read next time (lazy erase of other parts of the blob). However, getting just length would return valid value as blob index would most likely be present (being written after blob chunks).

negativekelvin, 2019-04-06:
Working with this patch 

```
esp_err_t Storage::eraseItem(uint8_t nsIndex, ItemType datatype, const char* key)
{
    if (mState != StorageState::ACTIVE) {
        return ESP_ERR_NVS_NOT_INITIALIZED;
    }

    if (datatype == ItemType::BLOB) {
        return eraseMultiPageBlob(nsIndex, key);
    }

    Item item;
    Page* findPage = nullptr;
    auto err = findItem(nsIndex, datatype, key, findPage, item);
    if (err != ESP_OK) {
        return err;
    }
    
    if (item.datatype == ItemType::BLOB_DATA || item.datatype == ItemType::BLOB_IDX) {
        return eraseMultiPageBlob(nsIndex, key);
    }

    return findPage->eraseItem(nsIndex, datatype, key);
}
```

```
I (376) NVS_EXAMPLE: nvs_open success
I (376) NVS_EXAMPLE: nvs_set_blob success
I (376) NVS_EXAMPLE: nvs_commit success
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=95 used=44 erased=51
 92: W ns= 4 type=66 span=  2 key="data" chunkIdx=0 len=8
 93: D
 94: W ns= 4 type=72 span=  1 key="data" chunkIdx=255 len=-1
I (856) NVS_EXAMPLE: nvs_open success
I (856) NVS_EXAMPLE: some_data nvs_get_blob success
source some_data[0] = 42
 source some_data[1] = 64
getting_some_data[0] = 42
getting_some_data[1] = 64
I (876) NVS_EXAMPLE: nvs_open success
I (886) NVS_EXAMPLE: nvs_erase_key success
I (886) NVS_EXAMPLE: nvs_commit success
state=fffffffe (ACTIVE) addr=a000 seq=1
firstUsed=0 nextFree=95 used=41 erased=54
 92: X
 93: X
 94: X
I (1826) NVS_EXAMPLE: Error (ESP_ERR_NVS_NOT_FOUND) All good - no NVS_KEY_DATA in nvs memmory
```

closed 2019-04-17 by igrr