espressif/esp-idf #3256: Multiple services with the same UUID using attribute tables

frazan-un, 2019-04-04:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.1.1-1-g70f4e5e62
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 3.3V

## Problem Description
Hi all,
I'm trying to make a profile with 2 services with the same UUID (same service table) using the attribute tables.
The function that create that (esp_ble_gatts_create_attr_tab()) is ok because there is an argument srvc_inst_id, and so I can call it 2 times with different service instances. My problem is that in the event generated (ESP_GATTS_CREAT_ATTR_TAB_EVT) I can't find a reference to the service instance in order to save the handles in different places: gatts_add_attr_tab_evt_param contains only a esp_bt_uuid_t not a esp_gatt_srvc_id_t like the gatts_create_evt_param (ESP_GATTS_CREATE_EVT) where instance ID can be found.
Is there a way to resolve this or can't be done using attr tables?
Thanks!

### Expected Behavior
```cpp
    /**
     * @brief ESP_GATTS_CREAT_ATTR_TAB_EVT
     */
    struct gatts_add_attr_tab_evt_param{
        esp_gatt_status_t status;       /*!< Operation status */
        esp_gatt_srvc_id_t service_id;  /*!< Service id, include service uuid and other information */
        uint16_t num_handle;            /*!< The number of the attribute handle to be added to the gatts database */
        uint16_t *handles;              /*!< The number to the handles */
    } add_attr_tab;                     /*!< Gatt server callback param of ESP_GATTS_CREAT_ATTR_TAB_EVT */

```
### Actual Behavior
```cpp
    /**
     * @brief ESP_GATTS_CREAT_ATTR_TAB_EVT
     */
    struct gatts_add_attr_tab_evt_param{
        esp_gatt_status_t status;       /*!< Operation status */
        esp_bt_uuid_t svc_uuid;         /*!< Service uuid type */
        uint16_t num_handle;            /*!< The number of the attribute handle to be added to the gatts database */
        uint16_t *handles;              /*!< The number to the handles */
    } add_attr_tab;                     /*!< Gatt server callback param of ESP_GATTS_CREAT_ATTR_TAB_EVT */

```

negativekelvin, 2019-04-04:
A workaround would be turn it into a synchronous call meaning do not call esp_ble_gatts_create_attr_tab the second time until you receive ESP_GATTS_CREAT_ATTR_TAB_EVT from the first call.

frazan-un, 2019-04-05:
Good idea! Thank you. I'll follow this way at the moment, however I think the modification should be introduced in the IDF for completeness.

Weijian-Espressif, 2019-04-13:
@frazan-un, Thanks for your reminder,  I will modify the IDF.

Weijian-Espressif, 2019-06-19:
@frazan-un , The commit is ceb00f7a4190e47b912444d2829575602d9b2a28,  please try, thanks.

Alvin1Zhang, 2019-06-19:
@frazan-un Thanks for reporting the issue, feel free to reopen if the issue still exists. Thanks.

closed 2019-06-19 by Alvin1Zhang