espressif/esp-idf #3257: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT never called

tbitowt, 2019-04-04:
Hi,

I'm having problem with continous starting gap scan. I would like to scan for 30s, then 30s of break.  The problem is that after calling `esp_ble_gap_start_scanning(30)` I expect that the scan would last for 30s, then the `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` event comes. Unfortunately the event is not fired.

Is there something else I need to configure?

My current code for initialization is:

```
esp_ble_scan_params_t BleScanner::bleScanParams = {
        .scan_type = BLE_SCAN_TYPE_ACTIVE,
        .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
        .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval = 0x50,
        .scan_window = 0x30,
        .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE};

void BleScanner::init(int scanDuration, int breakDuration)
{
    ESP_LOGI(TAG, "Init");

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    auto ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_ble_gap_register_callback(BleScanner::gapCallback);
    if (ret) {
        ESP_LOGE(TAG, "%s gap register failed, error code = %x\n", __func__, ret);
        return;
    }

    ret = esp_ble_gap_set_scan_params(&BleScanner::bleScanParams);
    if (ret) {
        ESP_LOGE(TAG, "%s gap set scan params failed, error code = %x\n", __func__, ret);
        return;
    }
    ESP_LOGV(TAG, "Initialization done");
}

void BleScanner::gapCallback(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
    ESP_LOGI(TAG, "Event called: %d", event);
    switch (event) {
        case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
            esp_ble_gap_start_scanning(scanDuration);
        }
        ...
    }
}
```

the only events I receive are:

```
I (686) BleScanner: Event called: 2
I (686) BleScanner: Event called: 7
I (746) BleScanner: Event called: 3
I (806) BleScanner: Event called: 3
I (1056) BleScanner: Event called: 3
... only event n. 3 till the end of scan duration
```

chegewara, 2019-04-04:
I cant say for sure its issue or not, but `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` is fired when you implicitly stop ble scan with `esp_ble_gap_stop_scanning`, but i agree, it would be useful to have `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` fired when scanning stops after defined time.

dhrishi, 2019-04-04:
@tbitowt When the duration expires, the device will stop scanning and `ESP_GAP_BLE_SCAN_RESULT_EVT` event will be received with `scan_result->scan_rst.search_evt` populated with `ESP_GAP_SEARCH_INQ_CMPL_EVT`

Weijian-Espressif, 2019-04-08:
@tbitowt , When the duration expires, `ESP_GAP_SEARCH_INQ_CMPL_EVT` is is triggered.
When you call `esp_ble_gap_stop_scanning()`, `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT ` will be triggered.

Alvin1Zhang, 2019-05-20:
@tbitowt Thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-07-05:
Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks. @tbitowt

closed 2019-07-05 by Alvin1Zhang